Xinyuan-LilyGO/LilyGo-Modem-Series #272: Need Support for https_put() method in TinyGSM

jawwadcreation, 2025-05-02:
Currently, the TinyGSM library provides native methods for https_post() and https_get(), but there is no official support for https_put().

This creates a challenge when working with services like Firebase Realtime Database, where we want to update/overwrite data at a specific path using the PUT method. Since https_put() does not exist, we are forced to use low-level manual AT commands, which are:

More complex to implement

Harder to debug and maintain

Less stable and consistent across networks

Adding native support for https_put() in TinyGSM would make the library more complete and easier to use in many real-world IoT scenarios.

Is there any support for https_put() ?

lewisxhe, 2025-05-02:
I will test it after the holidays and add

lewisxhe, 2025-05-02:
Already added ， example : https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/HttpsBuiltlnPut

carlestecno, 2025-05-04:
Thanks, @lewisxhe, 

It's working fine. 

I used a REST API (PHP) as an intermediate. Now, with HTTP PUT, I can send data directly to Firebase without needing the intermediary.

closed 2025-05-13 by lewisxhe