Xinyuan-LilyGO/LilyGo-Modem-Series #273: What's the difference between T-Call-A7670 V1.1 and V1.0?

Tsury, 2025-05-02:
I was working with 1.0, built a perfboard for it. I fried it, ordered a new one and got V1.1.
I see that the pins are different? Does it mean my perfboard is useless? Can I order the V1.0 still or is it gone?

lewisxhe, 2025-05-05:
Can you take photos of the front and back of the object?

lewisxhe, 2025-05-05:
V1.0 is the normal sales version, and you should buy V1.0 in the store.

lewisxhe, 2025-05-05:
This is a problem with the warehouse, they made a mistake. Contact the dealer and they will reissue you V1.0

Tsury, 2025-05-05:
@lewisxhe 
I see. Well, mistakes happen, thanks for clearing this up.

So I just contact the AliExpress store and show them this thread?

lewisxhe, 2025-05-05:
Yes

Tsury, 2025-05-05:
Thank you for the quick help @lewisxhe.

closed 2025-05-05 by Tsury