Xinyuan-LilyGO/LilyGo-Modem-Series #275: [Bug]:

IOTESLA-ing, 2025-05-05:
### Category

MQTT

### Hardware

T-SIM7670G S3

### PN Number

S2-10D1Y-Z32E5

### Firmware Version

```Shell
AT

OK

AT+SIMCOMATI

Manufacturer: SIMCOM INCORPORATED
Model: SIM7670G-MNGV
Revision: 2374B03SIM767XM5A_M
SIM767XM5_B03V02_240914
IMEI: 864643060153709
```

### Description

Hello everyone,

I previously purchased the A7670G model (https://lilygo.cc/products/t-sim-a7670e?srsltid=AfmBOoqm_LDQ2MeBE5tgXKTPP12HpZSoBLDc05Tb_46rd5tARXK_DdSV), and I had no issues at all—it worked perfectly.

Later, I bought the new SIM7670G model (https://lilygo.cc/products/t-sim-7670g-s3?_pos=1&_sid=65ae47a3b&_ss=r), but I'm having trouble loading the certificates required for establishing an MQTT connection.

For testing, I used the same example code MqttsBuiltInAWS.ino, only changing the device type in utilities.h by setting:

#define LILYGO_T_A7670    // A7670G
// or
#define LILYGO_T_SIM7670G_S3

The certificates belong to the same AWS Thing. I successfully published the test data using the A7670 example code, and I also verified the certificates using the Paho MQTT Python library to ensure the credentials are valid. So, I believe the issue is not related to AWS itself.

Has anyone experienced similar problems with the SIM7670G model, or do you have any suggestions on how to debug this?

LOG A7670G:
[LOG A7670.txt](https://github.com/user-attachments/files/20049387/LOG.A7670.txt)

LOG SIM7670G_S3 :
[LOG SIM7670G_S3.txt](https://github.com/user-attachments/files/20049388/LOG.SIM7670G_S3.txt)

### Relevant log output

```Shell
Start Sketch
Start modem...
AT
*   AT

QCRDY
AT
OK

AT+CPIN?
AT+CPIN?
+CME ERROR: SIM busy

+CPIN: READY

SMS DONE
SIM card online
Wait for the modem to register with the network.AT+CREG?
AT+CREG?
+CREG: 0,6

OK
Registration Status:6

Registration Status:6
AT+CPSI?
AT+CPSI?
+CPSI: LTE,Online,730-01,0x13F0,11621891,8,EUTRAN-BAND7,3100,5,5,-10,-113,-103,11

OK
Inquiring UE system information:LTE,Online,730-01,0x13F0,11621891,8,EUTRAN-BAND7,3100,5,5,-10,-113,-103,11
AT+NETOPEN
AT+NETOPEN
OK

+NETOPEN: 0AT+IPADDR

AT+IPADDR
+IPADDR: 10.120.36.131

OK
Network IP:AT+IPADDR
+IPADDR: 10.120.36.131
AT+CMQTTSTART
AT+CMQTTSTART
OK

+CMQTTSTART: 0
Connecting to a26mbzli1e5bxa-ats.iot.us-east-2.amazonaws.comAT+CCERTDOWN="ca_cert.pem",1189
AT+CCERTDOWN="ca_cert.pem",1189
>
-----BEGIN CERTIFICATE-----
MIIDQTCCAimgAwIBAgITBmyfz5m/jAo54vB4ikPmljZbyjANBgkqhkiG9w0BAQsF
ADA5MQswCQYDVQQGEwJVUzEPMA0GA1UEChMGQW1hem9uMRkwFwYDVQQDExBBbWF6
b24gUm9vdCBDQSAxMB4XDTE1MDUyNjAwMDAwMFoXDTM4MDExNzAwMDAwMFowOTEL
MAkGA1UEBhMCVVMxDzANBgNVBAoTBkFtYXpvbjEZMBcGA1UEAxMQQW1hem9uIFJv
b3QgQ0EgMTCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBALJ4gHHKeNXj
ca9HgFB0fW7Y14h29Jlo91ghYPl0hAEvrAIthtOgQ3pOsqTQNroBvo3bSMgHFzZM
9O6II8c+6zf1tRn4SWiw3te5djgdYZ6k/oI2peVKVuRF4fn9tBb6dNqcmzU5L/qw
IFAGbHrQgLKm+a/sRxmPUDgH3KKHOVj4utWp+UhnMJbulHheb4mjUcAwhmahRWa6
...
...
...
...
A4IBAQCY8jdaQZChGsV2USggNiMOruYou6r4lK5IpDB/G/wkjUu0yKGX9rbxenDI
U5PMCCjjmCXPI6T53iHTfIUJrU6adTrCC2qJeHZERxhlbI1Bjjt/msv0tadQ1wUs
N+gDS63pYaACbvXy8MWy7Vu33PqUXHeeE6V/Uq2V8viTO96LXFvKWlJbYK8U90vv
o/ufQJVtMVT8QtPHRh8jrdkPSHCa2XV4cdFyQzR1bldZwgJcJmApzyMZFo6IQ6XU
5MsI+yMRQ+hDKXJioaldXgjUkK642M4UwtBV8ob2xJNDd2ZhwLnoQdeXeGADbkpy
rqXRfboQnoZsG4q5WTP468SQvvG5
-----END CERTIFICATE-----

ERROR
Failed!
```

lewisxhe, 2025-05-06:
Please upgrade the firmware version to 2374B05SIM767XM5A_M

https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md

lewisxhe, 2025-05-06:
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/183

IOTESLA-ing, 2025-05-06:
@lewisxhe, Thanks. I can confirm that both MQTT over TLS work

[image]

closed 2025-05-07 by lewisxhe