Xinyuan-LilyGO/LilyGo-Modem-Series #276: Issue with SPK +/- (TTS-related) (T-A7670e)

p-sio, 2025-05-06:
Hello! I have a question regarding the SPK +/- at the back of the ESP32 board. I have tried the TTS sample code, and it works, but the volume is too low. Now, I tried this small amplifier:

[image]

I connected the "IN" to SPK +/- of the board; the VCC to the 3.3V pin and GND pin of the board (operating voltage: 3-5V); the OUT to a speaker but nothing happened. Now, I checked the continuity of the GND of this amplifier and they are connected, but when i tried checking just the lilygo board, I found that the GND pin from the board is not connected to the SPK- at the back of the board. Is it okay to connect the ground of the board to the SPK-?

I also looked up the SIMCON manual (A76XX_Series_AT_Command_Manual_V1.12) to see if there are useful commands. I tried this part under YOUNGTONE TTS page 467. The code compiled but nothing happened so I think it's not available in my board.

[image]

Can anyone give me some ideas to make the TTS louder? If possible, I don't want additional circuitry for the power source of an amplifier. Thank you!

lewisxhe, 2025-05-07:
SPK- cannot be connected to GND, otherwise the modem will be damaged.
Your amplifier board is single-ended input, IN- is directly connected to GND, you cannot connect SPK- to GND like this.

You can try to connect SPK- to any of the 2nd and 3rd pins of the chip IC for testing (may not work)

Or connect SPK- to IN- through a 1uF capacitor.

lewisxhe, 2025-05-07:
Figure 2 is correct.

p-sio, 2025-05-07:
I will try that. Thanks a lot!