Xinyuan-LilyGO/LilyGo-Modem-Series #277: Issue with T-A7608

namdohun, 2025-05-07:
Uploading an ATDebug example is causing the following error. Is there a solution to this?

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x29 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403795d5
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x118c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc20
load:0x403cb700,len:0x30e0
entry 0x403c88b8
Start Sketch

lewisxhe, 2025-05-07:
Which board did you choose? In [utilities.h](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/examples/ATdebug/utilities.h)

namdohun, 2025-05-08:
@lewisxhe I was choose this one. LILYGO_T_A7608X

lewisxhe, 2025-05-08:
Have you tested uploading a sketch that only prints helloworld on the serial? Is it the same problem?

If it is still the same, please shoot a video of uploading a simple sketch, such as a sketch that prints helloworld, and shoot a video of the serial repeatedly restarting and send it to the dealer, and they will handle the after-sales problem for you.

huynhdang-146, 2025-06-05:
Dear Sir,
I am using T-A7608EH. I would like to use it to test Hive 
MQTT application ( TLS/SSL port 8883). I use a code in example LilyGo T-A76xx. But it not successful. It can not connect to Hive MQTT ( MosquittoMQTT is ok)
How to test this? Please tell me something 

Thank you

lewisxhe, 2025-06-06:
I have tested all the sample codes in this repository. If you cannot connect, please open a new issue and paste the hardware name and modem version, and post all debug information.