Xinyuan-LilyGO/LilyGo-Modem-Series #278: A7670E-LNXY-UBL SLL and HTTPS problem

Bulbik1500, 2025-05-08:
----SIM MODE INFO----
AT+SIMCOMATI

 Manufacturer: SIMCOM INCORPORATED
 Model: A7670E-LNXY-UBL
 Revision: 23110B01A7670M4A_CUS_UBIA_91530228
 A7670M4_B01V01_230724
---------

for reference im using or tried using this following source: 
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/datasheet/A76XX
https://github.com/vshymanskyy/TinyGSM/tree/master/examples

i have problem with AT-commands for SSL or HTTPS , it seem like they are not supported beceause only respone i get from them is ERROR.
even examples from diffrent libraries don't work.
im able to connect to network but that's basically it.
i did see updating firmware might help but for a life of me i can't find it.  

On the side note i also have ttgo with sim800L which i updated to Revision:1309B10SIM800L16 and i question is it able to make SSL/TCL HTTPS connection.

lewisxhe, 2025-05-08:
Please test with examples from this repository..

Bulbik1500, 2025-05-08:
sorry that was bad communication on my part, i did use examples from their libraries,

[image]
[Network.txt](https://github.com/user-attachments/files/20099970/Network.txt)

this is what i get.

lewisxhe, 2025-05-08:
I don't have A7670E-LNXY-UBL now, as far as I know A7670E-LNXY-UBL and other versions should be the same, I need a few days, I need an A7670E-LNXY-UBL for testing.

Bulbik1500, 2025-05-08:
okay, i also have ants-make am-036 which is ttgo esp32 with sim800L and i find firmware something 14.... but i wasn't able to install it bc i get error about not correct chipset.

lewisxhe, 2025-05-09:
The problem has been found. This version of the firmware has a problem. Please update the built-in firmware of the modem according to this page.

Tips: The updated firmware version currently has a 715 error when accessing certain websites. I have communicated with simcom about this. But it can connect to the network and use it normally.

https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md

closed 2025-05-09 by Bulbik1500