Xinyuan-LilyGO/LilyGo-Modem-Series #279: Send payload from C#

cucilo, 2025-05-08:
I used:

- LILYGO_T-CALL A7670 v1.1 card
- AWS IoT
- C# app

When I sent payload from AWS IoT (MQTT test client), card work fine.
When I sent payload from C# app, card reset

`+CMQTTCONNLOST: 0,2
AT+CMQTTDISC?
AT+CMQTTDISC?
+CMQTTDISC: 0,1
+CMQTTDISC: 1,1`

In both situation, I send value 1, and into MQTT test client, I can saw message. It looks like the same.

[image]

This is C# code:

`      private readonly IMqttClient _mqttClient;
        private readonly string _awsEndpoint = "xxx-ats.iot.eu-west-3.amazonaws.com";
        private readonly string _clientId = "xxx";
        private readonly string _topic = "xxx/sub";
        private readonly string _certPath = "xxx.pfx";
        private readonly string _certPassword = "xxxx";

        public async Task ConnectAsync()
        {
            if (_mqttClient.IsConnected)
                return;

            var certificate = new X509Certificate2(_certPath, _certPassword);
            var rootCA = new X509Certificate2("xxx.pem");

            var options = new MqttClientOptionsBuilder()
                .WithTls(tls =>
                {
                    tls.UseTls = true;
                    tls.Certificates = new List<X509Certificate> { certificate, rootCA };
                    tls.CertificateValidationHandler = context =>
                    {
                        Console.WriteLine("TLS Validation: " + context.SslPolicyErrors);
                        return true; 
                    };
                    tls.AllowUntrustedCertificates = false;
                })
                .WithTcpServer(_awsEndpoint, 8883)
                .WithClientId(_clientId)
                .WithCleanSession()
                .Build();

            await _mqttClient.ConnectAsync(options, CancellationToken.None);
            Console.WriteLine("Connected to AWS IoT.");
        }

        public async Task<IActionResult> SendMessage_ESP(int id)
        {
            await ConnectAsync();

            var payloadObject = new { message = id.ToString() };
            string jsonPayload = System.Text.Json.JsonSerializer.Serialize(payloadObject);

            var mqttMessage = new MqttApplicationMessageBuilder()
                .WithTopic(_topic)
                .WithPayload(Encoding.UTF8.GetBytes(jsonPayload))
                .WithQualityOfServiceLevel(MQTTnet.Protocol.MqttQualityOfServiceLevel.AtLeastOnce)
                .Build();

            await _mqttClient.PublishAsync(mqttMessage);

            return Ok(new { status = "Message Sent", message = jsonPayload });
        }

`

lewisxhe, 2025-05-09:
I don't understand this, sorry.
Please try to test with a shared mqtt server

deeja, 2025-05-20:
@cucilo  - You will probably find more help here https://github.com/dotnet/MQTTnet 
AWS IoT is quite hard to start with due to the policies and certificates. 
I would suggest starting with Mosquitto MQTT server locally, or using https://www.hivemq.com/mqtt/public-mqtt-broker/

closed 2025-06-26 by cucilo