Xinyuan-LilyGO/LilyGo-Modem-Series #280: Can't put to sleep to A7670G

alexlopez2612, 2025-05-08:
Hello

I'm testing the ModemSleep example and my A7670G keep responding "AT" after the modem.sleepEnable(true).

This is the output of this sketch:

AT
*ATREADY: 1
AT
AT
OK
Modem is online !
Enter modem sleep mode!
AT+CSCLK=1
+CPIN: READY
SMS DONE
+CGEV: EPS PDN ACT 1
+CGEV: ME PDN ACT 8,1
AT+CSCLK=1
OK
Modem enter sleep modem!
Check modem response .
AT
AT
OK
.AT
AT
OK
..... and keeps forever like this....

Question 1: Why A7670G don't want to sleep?

I need to be able to sleep the modem and wake it up from:
1- A sms
2- By some command from the esp32

Question 2: Is this 2 ways possible?

Thanks!!

lewisxhe, 2025-05-09:
You didn't say which board.

alexlopez2612, 2025-05-09:
Hello @lewisxhe 
I tried both ways, connected and just with battery.
When I tried just with battery, the 2 modem red leds keeps on, and that's why I think that the modem is not sleeping.
Or, now I'm thinking, is normal that the leds remains on when the modem is in sleep mode?
Regards

alexlopez2612, 2025-05-17:
Hello @lewisxhe 
I did as you said and works ok.
But, to see what's happening, I need to be connectedo to serial to print traces. So, in I wanted to remove the resistor that you said, to make the A7670 go to sleep, but the image on the post that you sent me is not the same board that I have. mine is like this one:
https://lilygo.cc/cdn/shop/products/lilygo_3_96dfc126-9044-4f8f-8214-ec354e5a51be.jpg?v=1676363376&width=823
Please, can you tell me, on this board, wich one is the resistor that I have to remove?
Thank you!!!!!

alexlopez2612, 2025-05-19:
Thank you very much @lewisxhe! I remove the resistor and now the modem is sleeping like a baby connected to USB.
Just for confirmation, on the schematic, the resistor is RV7, right?
Again, thank you!!!!
Regards

alexlopez2612, 2025-05-20:
Hello @lewisxhe 
As always, thank you very much for your help.
Best regards!

closed 2025-05-20 by alexlopez2612