Xinyuan-LilyGO/LilyGo-Modem-Series #282: Which firmware to use for A7670E?

mariusebastian, 2025-05-10:
Hello

I'm following the modem firmware upgrade instructions, but the SIMCOMATI command does not return a model name. The instructions says to send the QR code on the modem to Lilygo so I'll put it here.

Which firmware do I use?

[image]

[image]

lewisxhe, 2025-05-10:
Did you not test which modem you have before upgrading?
I can’t reply to you today because I need to check the specific hardware version with SIMCOM.

mariusebastian, 2025-05-10:
I sent the simcomati command and the output is shown in the post above. Is
there anything else I can do to test which modem?

lewisxhe, 2025-05-12:
According to the PN number query, this is A7670E-FASE. Please update the firmware of A7670E-FASE.

sunnodetech, 2025-05-28:
Hello, I have this code, but I'm unable to connect to the API and retrieve the token.
When I run the following curl command:
`curl -X POST http://74.208.7.22:80/auth/token -H "Content-Type: application/json" -d '{"mac":"C8:2E:18:AD:5D:00"}'
`

{"token":"eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJtYWMiOiJDODoyRToxODpBRDo1RDowMCIsImlhdCI6MTc0ODM2MTY5OSwiZXhwIjoxNzUwOTUzNjk5fQ.dSPD5vHNtNCuIhRZMYr5_JdjXPEF6afWpvCP-F1J18U"}% 
`// Select your modem:
#define TINY_GSM_MODEM_SIM7600

// Set serial for debug console
#define SerialMon Serial

// Set serial for AT commands
#define SerialAT Serial1

// Enable verbose AT command dumping via StreamDebugger
#define DUMP_AT_COMMANDS 

// Define the serial console for TinyGSM's internal debug prints
#define TINY_GSM_DEBUG SerialMon 

// Your GPRS credentials
const char apn[]      = "data.lycamobile.fr";
const char gprsUser[] = ""; 
const char gprsPass[] = ""; 
#define GSM_PIN ""

// Server details for your API
const char* apiHost     = "http://74.208.7.22";
const uint16_t apiPort  = 80; 
const char* tokenEndpoint = "/auth/token";

#include <TinyGsmClient.h> 
#include <ArduinoJson.h>
#include <WiFi.h>

// Hardware Configuration
#define MODEM_RX 27
#define MODEM_TX 26
#define MODEM_PWRKEY 4
#define MODEM_BAUD 115200

// Global variables
String token = "";
String deviceMac = "";
bool tokenValid = false;
unsigned long lastTokenAttempt = 0;
const unsigned long tokenRetryCooldown = 60000;

// TinyGSM object
#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

// Function Prototypes
String sendATCommand(const String& cmd, unsigned long timeout = 2000, bool expectSpecificOK = true); 
bool isConnected(); 
void powerCycleModem();
bool setupModem();    
void establishConnection();
bool requestToken();  
void tokenManagement();
void getDeviceMac();
String readSerialResponse(unsigned long timeout = 1000, const char* terminator = nullptr);

void getDeviceMac() {
[... 500 lines cut ...]
void loop() {
  static unsigned long lastConnectionCheck = 0;
  const unsigned long connectionCheckInterval = 30000; 

  if (millis() - lastConnectionCheck > connectionCheckInterval) {
    lastConnectionCheck = millis();
    SerialMon.println("\n--- Periodic Check ---");
    if (!isConnected()) { 
      SerialMon.println("GPRS Connection lost! Attempting to reconnect...");
      tokenValid = false; 
      establishConnection(); 
      if(isConnected()){
         lastTokenAttempt = 0; 
         tokenManagement(); 
      } else {
         SerialMon.println("Failed to re-establish GPRS connection in periodic check.");
      }
    } else {
      SerialMon.println("GPRS Connection is stable.");
      tokenManagement(); 
    }
  }
  
  delay(1000); 
}`

closed 2025-07-12 by github-actions[bot]