Xinyuan-LilyGO/LilyGo-Modem-Series #283: Clarification on ESP32 & A7670E Sleep/Wake and GNSS Behavior T-SIM-A7670E.

Cjake28, 2025-05-11:
Context
I’m working on a low-power IoT tracker using an ESP32 paired with the LilyGO T-SIM-A7670E. Before diving into implementation, I’d like to understand the high-level behavior of both the ESP32 and the modem when they enter and exit sleep states.

Questions

1. Modem Sleep vs. MCU Sleep

When the A7670E enters its low-power sleep (PSM/eDRX or UART sleep), what happens to the ESP32 side? Can the microcontroller also deep-sleep without blocking wake-up signals?

2. GNSS Retention Across Sleep

If the modem is put into a sleep mode, does its GNSS engine remain powered and “hot,” or must it perform a full cold-start (re-download ephemeris) upon waking?

Is there an intermediate standby mode that preserves satellite data for faster fixes?

3. Network Re-Attachment

After waking, does the A7670E automatically re-register to the cellular network and resume data sessions seamlessly, or must the host re-issue APN/PDP context commands?

4. Wake-Up Triggers

While sleeping, can the modem detect incoming SMS or paging events and signal the ESP32 (e.g. via a GPIO interrupt) to wake the microcontroller?

What’s the recommended approach for using incoming network activity as a wake-up source?

Why This Matters
Understanding these states at a conceptual level will let me architect the firmware so that I can maximize battery life without sacrificing connectivity or GPS fix speed. Any diagrams, links to application notes, or design-pattern suggestions would be greatly appreciated!

lewisxhe, 2025-05-12:
1. When the modem goes to sleep, nothing will happen on the ESP side. The two are independent and each controls its own behavior.

2. In sleep mode, as long as the GNSS power is not turned off, the GNSS power is provided by the MODEM, which means that the modem cannot be turned off and can only be set to sleep state. It can maintain hot start. If the modem is turned off, it cannot be hot started.

3. Sleep wakeup does not need to be able to seamlessly resume data calls, because the sleep state does not disconnect the network connection

4. The modem can be awakened by ring and SMS in sleep state, but it cannot be awakened by network events. It does not have this function.

If you have high requirements for power consumption, SIM7670G is more suitable at present. I have made a SIM7670G tracking example. You can see it here.
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/tree/main/examples/Traccar

The functions you mentioned are basically tested with examples. Please check the examples. They basically contain my test descriptions.

Cjake28, 2025-05-17:
Hi @lewisxhe 
Sorry for the late reply — I’ve been really busy lately. Just wanted to say thank you for your helpful response. I really appreciate it!

closed 2025-07-01 by github-actions[bot]