Xinyuan-LilyGO/LilyGo-Modem-Series #285: [Bug]: 7076E Unable to connect to network

Aikufurr, 2025-05-12:
### Category

Other

### Hardware

T-A7670

### PN Number

S2-10D9P-Z325X

### Firmware Version

```Shell
14:59:26.086 -> AT+SIMCOMATI
14:59:26.119 -> Manufacturer: SIMCOM INCORPORATED
14:59:26.119 -> Model: A7670E-LNXY-UBL
14:59:26.119 -> Revision: 23110B01A7670M4A_CUS_UBIA_91530228
14:59:26.119 -> A7670M4_B01V01_230724
14:59:26.119 -> IMEI: 862771076562563
```

### Description

Using the MQTTSBuiltInAuth example with no modifications other than setting my APN (which works on my phone). I get `Enable network failed` when it attempts to use `setNetworkActive` I've tried auto and LTE settings to no luck, and all other network based examples get this error.

### Relevant log output

```Shell
15:01:45.196 -> AT+CPIN?
15:01:46.659 -> 
15:01:46.659 -> +CPIN: READY
15:01:46.659 -> AT+CPIN?

15:01:46.659 -> +CPIN: READY
15:01:46.659 -> 
15:01:46.659 -> OK
15:01:46.659 -> SIM card online
15:01:47.660 -> AT+CNMP=2
15:01:47.920 -> AT+CNMP=2
15:01:47.920 -> +CME ERROR: phone failure
15:01:48.659 -> Set network mode failed!
15:01:48.659 -> AT+CNMP?
15:01:48.659 -> AT+CNMP?

15:01:48.691 -> +CNMP: 38
15:01:48.691 -> 
15:01:48.691 -> OK
15:01:48.691 -> Current network mode : LTE
15:01:48.691 -> Set network apn : web
15:01:48.691 -> AT+CGDCONT=1,"IP","web"
15:01:48.691 -> AT+CGDCONT=1,"IP","web"

15:01:48.691 -> OK
15:01:48.691 -> Wait for the modem to register with the network.AT+CEREG?
15:01:48.691 -> 
15:01:48.691 -> +CGEV: EPS PDN ACT 1
15:01:48.820 -> AT+CEREG?
AT+CGREG?
15:01:49.948 -> 
15:01:49.948 -> +CEREG: 0,1
15:01:49.948 -> 
15:01:49.948 -> OK
15:01:49.948 -> Online registration successful
15:01:49.948 -> 
15:01:49.948 -> Registration Status:1
15:01:50.949 -> AT+CPSI?
15:01:50.949 -> AT+CGREG?

15:01:50.949 -> +CGREG: 0,1
15:01:50.949 -> 
15:01:50.949 -> OK
15:01:50.949 -> AT+CPSI?

15:01:50.981 -> +CPSI: LTE,Online,234-30,0x60AF,3513350,109,EUTRAN-BAND7,3350,5,0,18,49,50,4
15:01:50.981 -> 
15:01:50.981 -> OK
15:01:50.981 -> Inquiring UE system information:LTE,Online,234-30,0x60AF,3513350,109,EUTRAN-BAND7,3350,5,0,18,49,50,4
15:01:50.981 -> AT+NETOPEN
15:01:50.981 -> AT+NETOPEN

15:01:50.981 -> ERROR
15:01:51.976 -> Enable network failed!
```

lewisxhe, 2025-05-13:
See https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/278

Aikufurr, 2025-05-13:
Fixed my issue, thank you!

closed 2025-05-13 by Aikufurr