Xinyuan-LilyGO/LilyGo-Modem-Series #287: The board needs a large capacitor to support the modem power spikes

deeja, 2025-05-20:
I've had some A7670SA and G boards for approx 2 years, and I had found them very inconsistent in regards to connecting to the network, and very annoyingly they needed at least 3.8 volts to work correctly. 

Just recently I found some very low power Lorawan IoT devices had capacitors attached to the batteries. I did some research and found that adding a decent sized, low ESR, capacitor would help. 

The difference is significant! 
Below is some code that records the voltage and the result of a simple GET request.

```txt
3786,3784,3778,Failed to start modem <-- first power on. Modem needs a bit of time
3772,3758,3750,3692,3732,complete <-- successfully downloaded
3770,3764,3734,3742,3734,complete
3770,3750,3756,3734,3744,complete
3778,3762,3734,3736,3742,complete
3768,3758,3746,3738,3732,complete
3772,3768,3730,3718,3734,complete
3766,3760,3746,3742,3736,complete
3768,3760,3742,3734,3658,complete
3766,3762,3740,3744,3758,complete
3770,3762,3742,3724,3734,complete
3786,3776,3776,Failed to start modem <-- inconsistent failures
3764,3758,3740,3732,3734,complete
3766,3768,3730,3742,3732,complete
3784,3776,3786,Failed to start modem
3770,3758,3740,3738,3652,complete
3772,3746,3750,3736,3716,complete
3766,3760,3746,3744,3734,complete
3782,3772,3776,Failed to start modem
3760,3768,3762,3734,3738,complete
3746,3758,3746,3734,3734,complete
3754,3736,3746,3734,3732,complete
3776,3772,3774,Failed to start modem
3766,3756,3734,3734,3732,complete
3768,3752,3740,3740,3732,complete
3790,3772,3758,Failed to start modem
3758,3750,3748,3732,3714,complete
3768,3756,3738,3740,3726,complete
3778,3772,3782,Failed to start modem
3764,3752,3750,3734,3732,complete
3764,3754,3736,3720,3710,complete
3782,3770,3772,Failed to start modem
3760,3758,3740,3720,3732,complete
3758,3756,3736,3734,3740,complete
3762,3758,3746,3734,3732,complete
3770,3756,3746,3734,3682,complete
3762,3760,3746,3734,3732,complete
3762,3752,3752,3734,3570,complete
3758,3758,3754,3734,3728,complete
3760,3752,3742,3734,3732,complete
3764,3760,3750,3732,3732,complete
3762,3754,3718,3736,3734,complete
3758,3752,3744,3688,3724,complete
3756,3758,3734,3730,3732,complete
3758,3744,3742,3732,3732,complete
3762,3746,3734,3732,3724,complete
3758,3744,3744,3732,3734,complete
3752,3750,3732,3732,3734,complete
3758,3748,3750,3730,3728,complete
3758,3750,3734,3726,3730,complete
3762,3746,3734,3726,3664,complete
3758,3746,3734,3716,3660,complete
3758,3746,3742,3726,3718,complete
3772,3762,3766,Failed to start modem
3750,3750,3744,3734,3732,complete
3762,3750,3730,3734,3730,complete
3770,3770,3772,Failed to start modem
3750,3744,3736,3720,3630,complete
3756,3742,3740,3718,3722,complete
3766,3762,3772,Failed to start modem
3758,3750,3716,3726,3716,complete
3762,3746,3712,3716,3732,complete
3754,3750,3734,3700,3720,complete
3752,3752,3738,3720,3708,complete
3752,3744,3734,3724,3700,complete
3734,3740,3740,3728,3632,complete
3768,3762,3764,Failed to start modem
3756,3746,3734,3730,3724,complete
3758,3742,3736,3724,3732,complete
3746,3738,3730,3730,3724,complete
3760,3740,3734,3714,3714,complete
3752,3744,3734,3724,3706,complete
3746,3736,3734,3724,3716,complete
3750,3724,3730,3708,3702,complete
3746,3742,3736,3724,3708,complete
3750,3738,3732,3720,3720,complete
3750,3740,3714,3708,3724,complete
3750,3740,3724,3708,3716,complete
3746,3746,3708,3714,3716,complete
3762,3752,3754,Failed to start modem
3742,3730,3720,3694,3718,complete
3742,3738,3708,3710,3724,complete
3750,3744,3734,3704,3706,complete
3760,3752,3752,Failed to start modem
3744,3742,3724,3702,3714,complete
3746,3736,3730,3714,3702,complete
3744,3738,3734,3718,3714,complete
3742,3736,3720,3720,3708,complete
3742,3736,3728,3718,3712,complete
3744,3736,3732,3726,3710,complete
3750,3742,3734,3726,3714,complete
3732,3732,3720,3710,3716,complete
3754,3752,3752,Failed to start modem
3746,3736,3710,3704,3646,complete
3762,3746,3748,Failed to start modem
3742,3734,3734,3710,3648,complete
3734,3740,3708,3714,3712,complete
3744,3726,3710,3728,3708,complete
3740,3722,3724,3708,3710,complete
--------------------------- Added capacitor ------------------------------
3772,3742,3750,Failed to start modem <--- cold start so expected
3770,3758,3758,Failed to start modem
3760,3750,3732,3708,3694,complete  <--- no failures after this point
3752,3744,3734,3710,3718,complete
3756,3740,3734,3716,3708,complete
3750,3740,3720,3732,3726,complete
3746,3740,3732,3724,3724,complete
3746,3736,3732,3728,3718,complete
3744,3736,3714,3734,3718,complete
3750,3738,3720,3724,3718,complete
3744,3730,3734,3720,3708,complete
3746,3742,3726,3726,3724,complete
3750,3736,3728,3712,3708,complete
3742,3734,3732,3716,3730,complete
3742,3734,3734,3710,3716,complete
3750,3734,3732,3718,3710,complete
3746,3734,3732,3716,3716,complete
3742,3738,3716,3718,3712,complete
3744,3736,3732,3726,3708,complete
3736,3734,3710,3704,3634,complete
3744,3734,3708,3716,3724,complete
3734,3740,3732,3724,3716,complete
3742,3734,3708,3718,3734,complete
3750,3732,3708,3712,3718,complete
3736,3736,3724,3708,3708,complete
3740,3730,3718,3710,3700,complete
3736,3734,3730,3694,3714,complete
3738,3732,3724,3720,3710,complete
3734,3738,3724,3708,3716,complete
3734,3734,3708,3708,3708,complete
3740,3734,3724,3712,3718,complete
3746,3736,3712,3724,3734,complete
3734,3734,3728,3710,3694,complete
3742,3726,3718,3708,3694,complete
3736,3732,3716,3708,3708,complete
3734,3734,3714,3724,3698,complete
3742,3732,3718,3708,3704,complete
3734,3732,3706,3712,3706,complete
3734,3734,3718,3710,3684,complete
3716,3734,3718,3708,3562,complete
3734,3734,3714,3698,3706,complete
3740,3730,3700,3708,3708,complete
3734,3724,3708,3716,3708,complete
3736,3726,3714,3692,3622,complete
3734,3724,3716,3708,3708,complete
3734,3732,3706,3708,3714,complete
3732,3728,3716,3716,3708,complete
3734,3724,3700,3730,3734,complete
3734,3724,3720,3690,3708,complete
3734,3724,3720,3708,3708,complete
3734,3728,3698,3712,3708,complete
3730,3734,3702,3704,3704,complete
3732,3724,3708,3708,3682,complete
3728,3720,3708,3702,3702,complete
3730,3726,3714,3706,3704,complete
3730,3724,3710,3708,3630,complete
3726,3730,3710,3704,3708,complete
3726,3726,3714,3698,3698,complete
3724,3720,3716,3708,3708,complete
3732,3718,3708,3706,3708,complete
3732,3720,3710,3668,3708,complete
3728,3732,3706,3734,3704,complete
3726,3718,3708,3708,3682,complete
3726,3716,3714,3706,3604,complete
```
https://www.jaycar.co.nz/2200uf-10vdc-low-esr-electrolytic-capacitor/p/RE6306
[image]

I have also had issues with batteries failing quickly in these devices. I suspect that adding this capacitor will help with that too.

lewisxhe, 2025-05-20:
This board is really power hungry. Adding bulk capacitors really helps smooth out the power.
We have a revision of S3 now, I will try to fix the issues with the previous version. Thanks for the report.

deeja, 2025-05-20:
I think you should add a warning to the documentation that without a capacitor the board will not function correctly. 
I just tested a board from 2 years ago that I thought was broken, but is working fine with a capacitor. 

It is very frustrating to know adding a capacitor solves a big problem

lewisxhe, 2025-05-20:
This is not always the case. Have you tried changing the battery? There may be aging of some components.

deeja, 2025-05-20:
New batteries work where the voltage is over 3.9v.

The issue is that these devices get deployed in places where the batteries need to last for 2 or more years. 
The battery will eventually be old; the voltage will decrease and the internal resistance will increase. 
With this capacitor in place, I am getting stable functionality down to 3.4 volts with old batteries. 

Another benefit of this is better solar charging. It's much easier to charge a battery from 3.4 volts to 3.8, than from 3.9 to 4.1v
This means in the winter my devices will work for longer, and might not need to sleep overnight.

deeja, 2025-05-20:
Just so you know, I really like the boards you make. They are exactly what I needed for the projects I am working on. The ONLY reason I stopped using them was because of the inconsistent connections. Now that I know how to fix it, I will start to use them again :D which is good because I have a few of them lying around.

GTorreil, 2025-05-30:
@lewisxhe do you have an idea when the S3 revision will be available and if it will fix this issue plus the sleep hangs and brownout issues ?

lewisxhe, 2025-05-30:
The S3 version was designed and tested by me, and the known problems have been optimized. However, it will take some time to wait. I need some testing time. I will publish the schematic diagram of my design next week.
1. For the undervoltage problem, a better power supply solution was replaced
2. An external watchdog circuit was added to lock the ESP and prevent it from starting when it is lower than 2.8V
3. GPS was unlocked, and the MODEM GPS and PPS can be accessed directly through ESP.
4. eSIM support was added
5. External audio amplifier was added
6. OV series camera support was added
7. Compatible with all current T-SIM expansion boards
8. Maximize the reasonable use of GPIO

GTorreil, 2025-05-30:
> The new revision is discussed in this issue. I still need some time to do rigorous testing. [#293](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/293)

Sorry, I missed that. 👍