Xinyuan-LilyGO/LilyGo-Modem-Series #289: Overheating Issue on T-SIM7670G-S3 – TVS or Other Component Failing?

IOTESLA-ing, 2025-05-21:
Hello Community,

I need your help. I was working with the Saleae and my T-SIM7670G-S3, and it worked excellently without issues while developing the driver for a CO₂ sensor.

My problem began when, at some point, the green LED (named that way in the schematic and visually the same) and the red LED (same case) started blinking intermittently. I ignored it and disconnected everything.
It worked fine afterward, but then the same issue returned, and later an EDx component (name from the schematic, marked as 1) started heating up, which was alarming.
I disconnected the board, visually inspected the components, and nothing looked damaged. Then I reconnected it, and that TVS (mark 1) stopped heating up. I checked it by touching it—yes, it was risky, but I needed to verify whether it was truly faulty.
Later, another section—sector 2—started heating up.

Has this happened before?
Which component should I replace?

[image]

Thanks u for the help!

lewisxhe, 2025-05-22:
Image position 1 is the power selection circuit. Does the heating occur when the USB is connected or when the battery is powered?

Image 2 is the boost part. Does it heat up when using USB or battery?

Caanee, 2025-05-23:
I experienced two different situations, both happening sequentially in time:

Initially, in position 1 (the power selection circuit), there was noticeable heating when I connected the USB for the first time. I found it odd, so I unplugged and reconnected the USB. On the second attempt, position 1 no longer heated up.

Later on, position 2 (the boost converter area) began heating up—also when using USB. Since then, only position 2 heats up when USB is connected.

Current behavior:

With USB: Position 2 heats up, and the green LED turns on.
With battery: Nothing happens—no heating, and even the green LED stays off.

I manually checked the temperature by touch and confirmed that the heat is concentrated around the resistors labeled "100" and the nearby capacitor in position 2.

IOTESLA-ing, 2025-05-23:
I experienced two different situations, both happening sequentially in time:

Initially, in position 1 (the power selection circuit), there was noticeable heating when I connected the USB for the first time. I found it odd, so I unplugged and reconnected the USB. On the second attempt, position 1 no longer heated up.

Later on, position 2 (the boost converter area) began heating up—also when using USB. Since then, only position 2 heats up when USB is connected.

Current behavior:

With USB: Position 2 heats up, and the green LED turns on.
With battery: Nothing happens—no heating, and even the green LED stays off.

I manually checked the temperature by touch and confirmed that the heat is concentrated around the resistors labeled "100" and the nearby capacitor in position 2.

lewisxhe, 2025-05-28:
The voltage is correct. It looks like it's working fine. Is it warm now?

IOTESLA-ing, 2025-05-28:
Yes, when I connect it via USB, the area I mentioned as zone 2 gets very hot. Also, there's now a strange sound, like a knocking or tapping from some component. It's also worth noting that no COM port is recognized when I plug in the USB, which suggests to me that the ESP32 isn't getting voltage or something unusual is happening.

closed 2025-07-13 by github-actions[bot]