Xinyuan-LilyGO/LilyGo-Modem-Series #293: New Revision Discussion

lewisxhe, 2025-05-30:
This issue discusses opinions on the hardware revision of the S3 version. Welcome to discuss

The S3 version is designed and tested by me, and the currently known problems are optimized.

1. For the undervoltage problem, a better power supply solution is replaced

2. An external watchdog circuit is added to lock the ESP and prevent it from starting when it is lower than 2.8V

3. Unlocked GPS, you can directly access the MODEM GPS and PPS through ESP.

4. Added eSIM support

5. Added external audio amplifier

6. Added OV series camera support

7. Compatible with all current T-SIM expansion boards

8. Maximize the reasonable use of GPIO

9. Added GPS backup battery to maintain GPS ephemeris data, so as to achieve hot start GPS

chemmex, 2025-06-30:
Do you mind changing the CN3165 to an MPPT capable charger? Like CN3791 or maybe CN3795 to use other battery chemistries? 
I myself prefer to use LiFePo4 batteries to prevent degradation in cold conditions, so this makes sense.

lewisxhe, 2025-06-30:
I have considered CN3791, but the PCB area is limited. CN3795 requires more peripheral components, and this PCB can no longer accommodate more components.
The PCB is already densely packed.

chemmex, 2025-06-30:
By the way, can you comment why there are two cascaded 3.3V converters? It seems that SY8089 is just bypassed by the D8. Is there a special reason for that?

lewisxhe, 2025-08-02:
It's not finished yet, I won't be able to get the latest version to test until the day after tomorrow. But it should be soon.

lewisxhe, 2025-08-07:
[image]
The new version has been prepared and is now being arranged for production. It is strongly recommended to use the new version. I will add relevant documents to explain it later. The main feature of this series is compatibility. All versions have the same IO, only different modems. At the same time, there is a seamless switching function, and it can also work at ultra-low voltage.

GTorreil, 2025-08-10:
This is great news @lewisxhe . Please keep us posted with the link to get the new units. I will try one in a real-world senario to see how it handles.