Xinyuan-LilyGO/LilyGo-Modem-Series #294: I have a LILYGO® TTGO T-A7670G/E/SA Model: A7670E-LNXY-UBL and i can not use Internet

davidfritsch1965, 2025-06-02:
I want to use my Lilygo to send sensor values to thingspeak.

I can do 
AT+CRESET
OK

*ATREADY: 1
+CPIN: READY
+CGEV: EPS PDN ACT 1

at+cgdcont?
+CGDCONT: 1,"IP","online.telia.se.mnc001.mcc240.gprs","10.43.156.248",0,0,,,,

AT+CGPADDR=1
+CGPADDR: 1,10.35.123.95

AT+CREG?
+CREG: 0,1

AT+CPIN?
+CPIN: READY

But when i try to 
at+netopen
ERROR
AT+CIPSHUT
ERROR
AT+IPADDR
ERROR
AT+PING="8.8.8.8"
ERROR
AT+CHTTPCREATE="httpbin.org",80
ERROR
AT+CHTTPCREATE=?
ERROR

For me it seems that my lilygo will not do any internet commands at all.

Information about the board
AT+SIMCOMATI

Manufacturer: SIMCOM INCORPORATED
Model: A7670E-LNXY-UBL
Revision: 23110B01A7670M4A_CUS_UBIA_91530228
A7670M4_B01V01_230724
IMEI: 862771076563850

Can i get help?
Is something wrong ?
Can i upgrade ?

lewisxhe, 2025-06-03:
See https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/278

davidfritsch1965, 2025-06-04:
I am on a ubuntu laptop. Do you have a way to upgrade firmware when running ubuntu

lewisxhe, 2025-06-05:
There is no upgrade software for liunx, it can only run on win

davidfritsch1965, 2025-06-06:
I managed to borrow a windows computer 

I dont manage to do the firmware update, the update tool keep ending up on " getting serial devices list "

[image]

lewisxhe, 2025-06-07:
It is plugged into the MicroUSB port on the motherboard.