Xinyuan-LilyGO/LilyGo-Modem-Series #299: [Bug]: AT+CGNSINF returns ERROR

sbond-solx, 2025-06-13:
### Category

GNSS

### Hardware

T-A7670

### PN Number

S2-10DAX-Z32C4

### Firmware Version

```Shell
Model: A7670E-FASE
 Revision: A110B01A7670M7_F
 A7670M7_B01V01_240105
 IMEI: 862771076818015
```

### Description

AT+CGNSINF returns ERROR.

### Relevant log output

```Shell
11:41:15.455 -> Serial AT bridge started.
11:41:15.455 -> Type AT commands below:
11:41:17.697 -> 
11:41:17.697 -> *ATREADY: 1
11:41:19.154 -> 
11:41:19.154 -> +CPIN: READY
11:41:21.582 -> 
11:41:21.582 -> +CGEV: EPS PDN ACT 1
11:41:22.221 -> 
11:41:22.221 -> SMS DONE
11:41:22.754 -> 
11:41:22.754 -> +CGEV: ME PDN ACT 8,1
11:41:25.615 -> 
11:41:25.615 -> PB DONE
11:41:33.022 -> AT
11:41:33.022 -> OK
11:41:36.004 -> AT+CGNSSPWR=1
11:41:36.116 -> OK
11:41:43.897 -> 
11:41:43.897 -> +CGNSSPWR: READY!
11:42:35.440 -> AT+CGNSINF
11:42:35.440 -> ERROR
11:45:16.993 -> AT+SIMCOMATI
11:45:16.993 -> Manufacturer: SIMCOM INCORPORATED
11:45:16.993 -> Model: A7670E-FASE
11:45:16.993 -> Revision: A110B01A7670M7_F
11:45:17.040 -> A7670M7_B01V01_240105
11:45:17.040 -> IMEI: 862771076818015
11:45:17.040 -> 
11:45:17.040 -> OK
11:47:42.342 -> AT+CGNSINF
11:47:42.342 -> ERROR
11:57:12.369 -> AT+CGNSINF
11:57:12.369 -> ERROR
```

sunnodetech, 2025-06-16:
maybe this is the solution : [https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/278](url)

lewisxhe, 2025-06-16:
Follow this issue for troubleshooting https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/137

sbond-solx, 2025-06-16:
I did some trial and error, and I was able to make it show the info that I needed by using your example sketch "GPS_BuiltInEx". I'd just like to ask how it works because I don't want to rely on the libraries you defined. Can you explain the workings of the code using AT commands?

lewisxhe, 2025-06-17:
If you don't want to rely on this library, you must understand the function of each command and the conditions before execution. The manual explains it. It is not advisable to execute the next command without waiting for the modem to return whether it is normal.

Especially when turning on GPS, you did not wait for GPS to turn on before executing GPS related commands. Please check the AT command manual for detailed instructions.

sbond-solx, 2025-06-17:
I'm reading the manual right now while trying to test the GPS using both your example "GPS_BuiltInEx" and trying AT commands. However, the AT commands aren't able to show me the location unlike your example, it mostly prints out

AT+CGNSSINF
,,,,,,,,,,,,,,,,,

closed 2025-07-31 by github-actions[bot]