Xinyuan-LilyGO/LilyGo-Modem-Series #305: Modem SIM7670G S3 don't start Power LED

Caanee, 2025-06-23:
Hello,  
I encountered an issue after upgrading the AT modem firmware to version **B05V01**, which was the intended goal.  
I’ve successfully upgraded several T-SIM7670 S3 units before, but this is the first one showing this problem.

The red power/startup LED on the modem never turns on either.
I checked **pin 12**, and it is **HIGH**, as expected.
However, in the code I get the following output:

`Start Sketch
Trying baud rate 115200
Trying baud rate 9600
Trying baud rate 57600
Trying baud rate 38400
Trying baud rate 19200
Trying baud rate 74400
Trying baud rate 74880
Trying baud rate 230400
Trying baud rate 460800
Trying baud rate 2400
Trying baud rate 4800
Trying baud rate 14400
Trying baud rate 28800
***********************************************************
 Failed to connect to the modem! Check the baud and try again.
***********************************************************`

I tried re-uploading the firmware, but the modem **doesn’t boot**.  
Interestingly, when I press the **BOOT button** on the AT modem, I **do see the 3 COM ports**, so I assume the modem is working at some level — but I’m not sure how to solve this.

Any ideas?

lewisxhe, 2025-06-23:
If the red light on the modem is not on, it may be that the update has failed. Can I try the upgrade again?

lewisxhe, 2025-06-24:
If the upgrade to B05 fails to start, try downgrading to B03/or other versions to see if it can start.

On the ESP32 side, you must make sure that the modem has been started in the program. The best way is to burn ATDebug for upgrading

closed 2025-08-08 by github-actions[bot]