Xinyuan-LilyGO/LilyGo-Modem-Series #311: [Bug]:

abcaste1006, 2025-06-27:
### Category

Other

### Hardware

T-A7670

### PN Number

S2-109ZY-Z32CK

### Firmware Version

```Shell
can't send at commands
```

### Description

Suddenly, I can no longer send AT commands or use the modem. For several days before, the board was able to be used normally via a USB-C connection. Now, it won't connect despite trying all baud rates. I disconnected all the GPIO pins, meaning nothing was connected, and the problem persists. All other functions (Wi-Fi, Telegram, etc.) work fine.

### Relevant log output

```Shell
Failed to connect to the modem! Check the baud and try again
```

lewisxhe, 2025-06-28:
1. Connect the microUSB cable to the board and select the AT port, provided that you have installed the driver. If not, please refer to the firmware upgrade guide to install the driver.

2. Change the serial monitor baud rate to 115200 and send AT to see if the data is returned normally.

3. If it is correct, send AT+IPR? to query the current baud rate setting

lewisxhe, 2025-06-28:
看起来modem工作正常，问题可能出现在电平转换电路 。
按照下面步骤进行测试。

1. 写入ATDebug ， 波特率固定 115200 ，移除 AutoBaud函数
2. 等待modem红灯亮起
3. 发送AT 查看是否正常返回OK

如果还是无法返回消息，请联系经销进行售后处理 ，请按照上述步骤拍摄一个视频发送给经销商，作为售后依据。

closed 2025-08-13 by github-actions[bot]