Xinyuan-LilyGO/LilyGo-Modem-Series #320: FTP(S) Connect socket failed after AT+CFTPSPUT in T-A7670E

teamlead-wit, 2025-07-15:
I have tried a log to put a file on FTP(S) server with difference sequence of AT command, however, I get Connect Socket Failed error when in Implicit Mode and only create a blank file on FTP(S) serer when in explicit mode. 
I can read/write/delete files using FileZilla, however, the issue only occurs when using GSM modem. 
i can login, create directory, change, delete directory, but only cannot LIST and PUT. 

AT+CSSLCFG="sslversion",3,4
AT+CSSLCFG="authmode",3,1
AT+CSSLCFG="cacert",3,"ftps_cert.der"
AT+CFTPSSLCFG=0,3
AT+CFTPSSLCFG=1,3
AT+CFTPSSTART
AT+CFTPSLOGIN="ftps_server_IP",990,"user","Password",3
AT+CFTPSMODE=0 // (tired both implicit and explicit mode here)
AT+CFTPSPWD // (works fine)
AT+CFTPSMKD="test" //(works fine)
AT+CFTPSRMD="test" //(works fine)

AT+CFTPSPUT="hello.txt" //here i get error
//if implicit mode  +CFTPSPUT: 14 and if explicit mode  +CFTPSPUT: 1

I have followed the application notes of A7670E, but all failed.

lewisxhe, 2025-07-15:
Where is your file? Did you put the file in the efs file system inside the modem?
And you don't have a path.

The correct procedure is to write the file to the modem efs file system
and then write it to the correct path

teamlead-wit, 2025-07-15:
Even if try AT+CFTPSLIST="/" and get +CFTPSLIST: 1. 
I am following the application notes uploaded on SIM Com website. Just LIST dir even does not work.

lewisxhe, 2025-07-16:
I need some time to test it. I will test it in a few days. I am going to add an example of FTP, but I don't have time these days. I will add an example of this this Saturday and Sunday. Please wait.

teamlead-wit, 2025-07-16:
I am able to read and write files on free FTP server.

AT+CFTPSSTART
AT+CFTPSLOGIN="ftp.drivehq.com",21,"user_name","password",0
AT+CFTPSPWD
AT+CFTPSGETFILE="readme.txt",1
AT+CFTPSGETFILE="test.csv",1
AT+CFTPSLOGOUT
AT+CFTPSSTOP

however the issue is with FTP(S). I think there could be any issue with certificate. I have tried to upload the certificate using AT+CCERTDOWN="name.der",size
but this command do work sometime and sometime error. I think I am not doing it correctly. 

What is the best way to upload the certificate? After uploading I set the context and do the following.

AT+CSSLCFG="sslversion",3,4
AT+CSSLCFG="authmode",3,1
AT+CSSLCFG="cacert",3,"ftps_cert.der"

AT+CFTPSSLCFG=0,3
AT+CFTPSSLCFG=1,3

teamlead-wit, 2025-07-17:
Can there be an issue with Self-Signed certificate?

teamlead-wit, 2025-07-26:
Issue resolve. There were restrictions on ftp server at client end. He has change the rules and it working fine as plain FTP. Furthermore, I have deployed my own FTP(S) and it works fine. I have not enabled the strict certificate requirement. Hence, no need for certificate.