Xinyuan-LilyGO/LilyGo-Modem-Series #323: HTTPS requests not working, keeps reverting to HTTP/1.1

PepeAnto, 2025-07-19:
# Introduction
I am trying to use the A7670E to make an HTTPS GET request. 
I actually want to make an HTTPS POST request later with GPS data to my own backend, but for now I just want to get the HTTPS GET request working. 
I have updated the firmware as detailed on this Github page, logs shown below.

## Original State
```bash
20:57:32.717 -> AT

20:57:32.717 -> OK
20:57:35.207 -> AT+SIMCOMATI

20:57:35.207 -> Manufacturer: SIMCOM INCORPORATED
20:57:35.207 -> Model: A7670E-FASE
20:57:35.207 -> Revision: A110B01A7670M7_F
20:57:35.207 -> A7670M7_B01V01_240105
20:57:35.207 -> IMEI: 8XXXXXXXXXXX994
20:57:35.207 -> 
20:57:35.207 -> OK
```

## Current State
[Firmware "A7670E-FASE GPS"](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md#a7670e-fase-gps)
```bash
19:24:48.745 -> AT

19:24:48.745 -> OK
19:24:50.436 -> AT+SIMCOMATI

19:24:50.436 -> Manufacturer: INCORPORATED
19:24:50.436 -> Model: A7670E-FASE
19:24:50.436 -> Revision: A011B07A7670M7_F
19:24:50.436 -> A7670M7_B07V01_240927
19:24:50.436 -> QCN:
19:24:50.436 -> IMEI: 8XXXXXXXXXXX994
19:24:50.436 -> MEID:
19:24:50.436 -> +GCAP: +CGSM,+FCLASS,+DS
19:24:50.436 -> DeviceInfo:
```
[A7670E_Program_Logs.txt](https://github.com/user-attachments/files/21329782/A7670E_Program_Logs.txt)

#  Method
I am using a python script that connects to the COM port created by the USB interface with the A7670E. This script prints the firmware and then tries to perform two HTTP(S) requests. The first is a normal HTTP request. The second is an HTTPS request to the same site. 

Take note that the response header states HTTP/1.1 instead of HTTP/2.
Another issue is that the body (**AT+HTTPREAD**) always returns ERROR.

```bash
>> AT+SIMCOMATI
<< AT+SIMCOMATI
<< Manufacturer: INCORPORATED
<< Model: A7670E-FASE
<< Revision: A011B07A7670M7_F
<< A7670M7_B07V01_240927
```

## Code
```bash
import serial
import time

# Configuration
SERIAL_PORT = 'COM14'  # Change to your port if needed
BAUD_RATE = 115200
TIMEOUT = 10  # seconds
APN = 'ws.air.mvno'  # Set your APN

# HTTP/HTTPS test URLs
HTTP_URL = 'http://httpbin.org/get'
HTTPS_URL = 'https://httpbin.org/get'
[... 277 lines cut ...]
>> AT+HTTPREAD
<< AT+HTTPREAD
<< ERROR

>> AT+HTTPTERM
<< AT+HTTPTERM
<< OK

>> AT+CGNSSPWR=1
<< AT+CGNSSPWR=1
<< OK
```

# Assumptions
The APN setting is correct (**AT+CGDCONT=1,"IP","ws.air.mvno"**) since we have an IP address assigned and we can do HTTP requests.
The correct firmware is installed according to kgmuzungu's [comment](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/117#issuecomment-2401405371) in his issue report.

# Notes
@lewisxhe I have read the issue logged in [HTTPS requests not working, error 715 #117](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/117) and I tried the Firmware mentioned kgmuzungu's [comment](https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/117#issuecomment-2401405371) but it still didn't work.
When I try to perform a GET request on my own backend I receive a 715 error. I have Cloudflare as a middle man between my server and any clients (A7670E). The Cloudflare SSL/TLS encryption mode is set to Full on my backend.

# Questions
1. Is my HTTP request function following the correct order of commands to be able to do HTTPS requests?
2. Can anyone supply logs of them making an HTTPS request where the response has HTTP/2 in it?
3. Are there other commands that must be sent to configure/initialise some SSL stuff in order to do HTTPS requests?

PepeAnto, 2025-07-21:
Thank you so much @lewisxhe, I tried your AT command sequence for "A011B07A7670M7_F" and it worked!
I modified the code to make an HTTPS POST to my backend and that also worked!

I created a python script to test it, here is the console output.

## Code
```python
import serial
import time

# Configuration
SERIAL_PORT = 'COM14'  # Change to your port if needed
BAUD_RATE = 115200
TIMEOUT = 10  # seconds

# HTTP/HTTPS test URLs
HTTP_URL = 'http://httpbin.org/get'
HTTPS_URL = 'https://httpbin.org/get'

# Open serial port
ser = serial.Serial(SERIAL_PORT, BAUD_RATE, timeout=TIMEOUT)

sensordata = True  # Set to True to mask IMEI and IP addresses

def send_command(cmd, delay=1):
    print(f"\n>> {cmd}")
    ser.write((cmd + '\r\n').encode('utf-8'))
    time.sleep(delay)
    return read_response()

def mask_number(s):
    digits = [c for c in s if c.isdigit()]
    if len(digits) < 2:
        return s
    first, last = digits[0], digits[-1]
    masked = first + 'X' * (len(digits) - 2) + last
    i = 0
    result = ''
    for c in s:
        if c.isdigit():
            if i == 0:
                result += first
            elif i == len(digits) - 1:
                result += last
            else:
                result += 'X'
            i += 1
        else:
            result += c
    return result

def mask_ip_line(line):
    if line.startswith('+CGPADDR:') or line.startswith('+CCHADDR:') or line.startswith('+IPADDR:'):
        if ',' in line:
            parts = line.split(',')
            if len(parts) >= 2:
                prefix = ','.join(parts[:-1])
                addr = parts[-1]
                addr_masked = '.'.join(mask_number(x) for x in addr.strip().split('.'))
                return prefix + ',' + addr_masked
        else:
            prefix, _, addr = line.partition(':')
            addr = addr.strip()
            if addr.count('.') == 3 and all(part.isdigit() for part in addr.split('.')):
                addr_masked = '.'.join(mask_number(x) for x in addr.split('.'))
                return prefix + ': ' + addr_masked
    return line

def read_response():
    lines = []
[... 515 lines cut ...]
            send_command('AT+CSSLCFG="enableSNI",0,1')
            send_command('AT+CSSLCFG="sslversion",0,4')
        send_command(f'AT+HTTPPARA="URL","{url}"')
        send_command('AT+HTTPACTION=0', delay=5) # 0 for GET
        send_command('AT+HTTPHEAD')
        
        lines = send_command('AT+HTTPREAD?')
        for line in lines:
            if '+HTTPREAD: LEN,' in line:
                try:
                    length = int(line.split(',')[2]) # Length is the third part for GET
                    if length > 0:
                        send_command(f'AT+HTTPREAD=0,{length}')
                except Exception:
                    pass
        send_command('AT+HTTPTERM')

    # Print device info at the end
    send_command('AT+SIMCOMATI')
    ser.close()
    print('--- Done ---')

if __name__ == '__main__':
    main()
```

lewisxhe, 2025-07-22:
Thanks for the python test script

closed 2025-09-06 by github-actions[bot]