Xinyuan-LilyGO/LilyGo-Modem-Series #328: help need

sumesh143, 2025-07-22:
i trying to connect barcode scanner UART and DS 3218 pro Digital servo in my LilyGO/LilyGO-T-A7670E 
on pin scaner RX 32 and TX 33 and servo 25 or 04 but it not working any help

lewisxhe, 2025-07-23:
This is the pin list of T-A7670, please do not use the already used GPIO
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/5c6fda039c9c4d2aba7b0efa59ae5d4a5a5c82be/examples/ATdebug/utilities.h#L62

[image]

lewisxhe, 2025-07-23:
This list shows that the A7670 board occupies these GPIOs. You must find unused GPIOs before you can use them, otherwise there will be conflicts.

lewisxhe, 2025-07-23:
Rx: 19 Tx 18

closed 2025-07-24 by sumesh143