Xinyuan-LilyGO/LilyGo-Modem-Series #329: External power supply

huynhdang-146, 2025-07-22:
Hi! I am using T-A7608EH. I would like to use external power supply without connecting USB-C. I have a Meanwell 5VDC output. Should i connect  the power 5VDC to VBUS pin GND pin? Should i use a capacitor in this case? 
Thank you

lewisxhe, 2025-07-23:
The connection to VBUS pin and GND pin is correct, no additional capacitor is needed, just make sure the power supply is stable

huynhdang-146, 2025-08-05:
Hi. I have a problem about turn on the power via USB-C.
I am testing for transferring data to MQTT. It is working well. So i test turn off the power and then turn on the power again
When I turn off the power (adapter 5VDC USB-C is removed), and then i plug the adapter again. It can't connect to MQTT. ( data can't transfer to MQTT server). So i removed the power and then plug to connect it again. It connect to MQTT. 
I don't know why. I try more. Have you ever seen  this case?

lewisxhe, 2025-08-05:
Is GPIO 12 set to high level?
[image]

lewisxhe, 2025-08-05:
Measure whether there is voltage at both ends
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/141#issuecomment-2350872488

lewisxhe, 2025-08-05:
No, just connect an 18650 battery and flip the switch to ON.

1. Measure whether gpio12 is high.
2. Measure the voltage across the two capacitors in the image linked above.

closed 2025-09-19 by github-actions[bot]