Xinyuan-LilyGO/LilyGo-Modem-Series #330: A7670G wont connect while roaming?

jesse377, 2025-07-23:
Im having trouble to receive and send text messages in other countries. It worked on the sim7070G but on this module it won't work. Is there a setting i need to change to use roaming?

jesse377, 2025-07-23:
Yes im sure. It is the same sim card as before. And it should be able to connect to 2g automatically right?

lewisxhe, 2025-07-24:
2G is OK. You should change the SIM card to test it.

jesse377, 2025-07-24:
I have checked and the sim does support 4g (according to the carrier). But if 4g connection is not possible shouldn't it automatically connect to 2g? Is there another reason it cannot send text messages outside the country?

lewisxhe, 2025-07-25:
It has nothing to do with MODEM_REG_SMS_ONLY. I want to see the registration status of CGREG

lewisxhe, 2025-08-19:
The factory default setting is Network Mode 2, which is automatic search mode. The A7670 does not have Network Mode 51.

If the search takes a long time, your signal may be very poor or you are on a specific frequency band. After the modem connects to the network, send AT+CPSI to check the network properties, verify your current network status, and then report it to me.

lewisxhe, 2025-08-19:
MODEM_REG_SMS_ONLY means only SMS messages are supported but no Internet or phone calls.

closed 2025-10-03 by github-actions[bot]