Xinyuan-LilyGO/LilyGo-Modem-Series #332: AT Commands not working with LILYGO T-CALL A7670 v1.1

saurabh-DE0, 2025-07-23:
Hello Sir, I have bought LILYGO T-CALL A7670 v1.1 for my aws mqtt project. I tried Awsbuiltinmqtt and ATdebug code also. But the AT commands are not working in serial terminal. I thought I have sim or module issue but when I called on that number the ring goes. Please help regarding it. I thought I got faulty device so, I bought another but still same issue with me. I tried to upgraded firmware using the USB pads of module still not fixed issue. Help me to resolve issue please. I tried many ways but nothing works. I have provide external power but it also doesn't work. In ATdebug code it just trying baud rate and says Failed to connect to the modem!

saurabh-DE0, 2025-07-24:
[image]

I have this exact device. And I choose  #define LILYGO_T_CALL_A7670_V1_0 in utilities.h

lewisxhe, 2025-07-24:
Is the silkscreen on your PCB V1.1 or V1.0?

lewisxhe, 2025-07-24:
OK. 1.0 and 1.1 have completely different pinouts. Please change to 1.1

saurabh-DE0, 2025-07-24:
Ok. I will try and let you know.
Thank you.

saurabh-DE0, 2025-07-24:
The AT commands are now working. I have tested the MqttsBuiltInAWS.ino code from example. I have just changed broker details to my details and I am using Airtel SIM with 'airtelgprs.com' APN. i also give all permission in AWS IOT policy as shown in your video. Also added all three files regarding certificate and key. But the MQTT isn't working. Check the following log and help to resolve issue.
**Log:**
23:13:27.192 -> AT
23:13:27.477 -> AT
23:13:27.758 -> AT
23:13:28.074 -> AT
23:13:28.212 ->
23:13:28.212 -> *ATREADY: 1
23:13:28.367 -> .
23:13:28.367 -> AT
23:13:28.367 -> AT

23:13:28.367 -> OK
23:13:28.367 ->
23:13:28.367 -> AT+CPIN?
23:13:28.367 -> AT+CPIN?
23:13:28.367 -> +CME ERROR: SIM failure
23:13:30.365 -> AT+CPIN?
23:13:30.400 -> AT+CPIN?
23:13:30.400 -> +CME ERROR: SIM failure
23:13:30.650 ->
23:13:30.650 -> +MSTK: 11, "D07B810301250082028182050C41697274656C20416C6572748F0E80437573746F6D657220436172658F1B814163746976617465204F746865722056415320536572766963658F1782446561637469766174652056415320536572766963658F058348656C708F0D924D7920426F6F6B6D61726B731F0600FFFFFFFF01"
23:13:30.858 ->
23:13:30.858 -> +CPIN: READY
23:13:31.032 ->
23:13:31.032 -> +MSTK: 11, "D00D8103010500820281829902090A"
23:13:31.066 ->
23:13:31.066 -> +MSTK: 14
23:13:31.803 ->
23:13:31.803 -> SMS DONE
23:13:31.872 -> SIM card online
23:13:32.858 -> AT+CNMP=2
23:13:32.858 ->
23:13:32.858 -> +CGEV: EPS PDN ACT 1
23:13:32.997 -> AT+CNMP=2

23:13:32.997 -> OK
23:13:32.997 -> AT+CNMP?
23:13:33.031 -> AT+CNMP?

23:13:33.032 -> +CNMP: 2
23:13:33.032 ->
23:13:33.032 -> OK
23:13:33.032 -> Current network mode : AUTO
23:13:33.032 -> Set network apn : airtelgprs.com
23:13:33.032 -> AT+CGDCONT=1,"IP","airtelgprs.com"
23:13:33.100 -> AT+CGDCONT=1,"IP","airtelgprs.com"

23:13:33.100 -> OK
23:13:33.100 -> Wait for the modem to register with the network.AT+CEREG?
23:13:33.285 -> AT+CEREG?

23:13:33.285 -> +CEREG: 0,1
23:13:33.285 ->
23:13:33.285 -> OK
23:13:33.285 -> Online registration successful
23:13:33.285 ->
23:13:33.285 -> Registration Status:1
23:13:34.266 -> AT+CPSI?
23:13:34.266 ->
23:13:34.266 -> +CGEV: ME PDN ACT 8,1
23:13:34.266 -> AT+CPSI?

23:13:34.266 -> +CPSI: LTE,Online,404-90,0x17E2,237060128,37,EUTRAN-BAND5,2425,2,0,10,46,45,7
23:13:34.266 ->
23:13:34.266 -> OK
23:13:34.266 -> Inquiring UE system information:LTE,Online,404-90,0x17E2,237060128,37,EUTRAN-BAND5,2425,2,0,10,46,45,7
23:13:34.303 -> AT+NETOPEN
23:13:34.303 -> AT+NETOPEN

[... 157 lines cut ...]
23:13:43.356 -> AT+CSSLCFG="authmode",0,2
23:13:43.356 -> AT+CSSLCFG="authmode",0,2

23:13:43.356 -> OK
23:13:43.356 -> AT+CMQTTREL=0
23:13:43.393 -> AT+CMQTTREL=0

23:13:43.393 -> +CMQTTREL: 0,20
23:13:43.393 ->
23:13:43.393 -> ERROR
23:13:43.393 -> AT+CMQTTACCQ=0,"A76XX",1
23:13:43.393 -> AT+CMQTTACCQ=0,"A76XX",1

23:13:43.393 -> OK
23:13:43.393 -> AT+CMQTTCFG="version",0,4
23:13:43.393 -> AT+CMQTTCFG="version",0,4

23:13:43.393 -> OK
23:13:43.393 -> AT+CMQTTCONNECT=0,"tcp://a1dtu0s2byy5cm-ats.iot.eu-north-1.amazonaws.com:8883",60,1
23:13:43.393 -> AT+CMQTTCONNECT=0,"tcp://a1dtu0s2byy5cm-ats.iot.eu-north-1.amazonaws.com:8883",60,1

23:13:43.712 -> +CMQTTCONNECT: 0,34
23:13:43.712 ->
23:13:43.712 -> ERROR
23:13:43.712 -> Failed!

lewisxhe, 2025-07-25:
Please check the firmware version. I have tested all the examples and there is no problem with A7670. https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md

saurabh-DE0, 2025-07-25:
My firmware is the same as the you mentioned in the comment log. Still it giving error

23:13:39.302 -> AT+SIMCOMATI

23:13:39.344 -> Manufacturer: INCORPORATED
23:13:39.344 -> Model: A7670E-FASE
23:13:39.344 -> Revision: A011B07A7670M7_F
23:13:39.344 -> A7670M7_B07V01_240927
23:13:39.344 -> QCN:
23:13:39.344 -> IMEI: xxxxxxxxxxxx
23:13:39.344 -> MEID:
23:13:39.344 -> +GCAP: +CGSM,+FCLASS,+DS
23:13:39.344 -> DeviceInfo:
23:13:39.344 ->
23:13:39.344 -> OK

saurabh-DE0, 2025-07-25:
Hello Sir,
MQTT with AWS successfully worked on my second device. This device has old firmware as following:
11:49:52.361 -> AT+SIMCOMATI
11:49:52.393 -> Manufacturer: SIMCOM INCORPORATED
11:49:52.425 -> Model: A7670E-FASE
11:49:52.425 -> Revision: A110B01A7670M7_F
11:49:52.460 -> A7670M7_B01V01_240105
11:49:52.492 -> IMEI: 862771076818585

But on my first device I have upgraded firmware to :
23:13:39.344 -> Revision: A011B07A7670M7_F
23:13:39.344 -> A7670M7_B07V01_240927

So now I want to downgrade this device to old firmware to test the working of it. So please provide me the old firmware link.
Thank you!

saurabh-DE0, 2025-07-25:
Thank you so much sir. With this firmware my another device is also working. Really appreciate you work and faster responses.

closed 2025-09-09 by github-actions[bot]