Xinyuan-LilyGO/LilyGo-Modem-Series #334: Urgent help SimA7670E

sumesh143, 2025-07-24:
My A7670E not connecting to sim data /

[image]
[image]

APN is du 
Internet provider is du dubai uae

lewisxhe, 2025-07-25:
What board?

sumesh143, 2025-07-25:
LILYGO® TTGO T-A7670E

[image]

lewisxhe, 2025-07-25:
What code did you use? How could there be SIM7000G in the log? Didn't you test the code of this repository?

sumesh143, 2025-07-25:
/*
  FILE: AllFunctions.ino
  AUTHOR: Koby Hale
  PURPOSE: Test functionality
*/

#define TINY_GSM_MODEM_SIM7000
#define TINY_GSM_RX_BUFFER 1024 // Set RX buffer to 1Kb
#define SerialAT Serial1

// See all AT commands, if wanted
#define DUMP_AT_COMMANDS

/*
   Tests enabled
*/
#define TINY_GSM_TEST_GPRS    true
#define TINY_GSM_TEST_GPS     true
#define TINY_GSM_POWERDOWN    true

// set GSM PIN, if any
#define GSM_PIN ""

// Your GPRS credentials, if any
const char apn[]  = "YOUR-APN";     //SET TO YOUR APN
const char gprsUser[] = "";
const char gprsPass[] = "";

#include <TinyGsmClient.h>
#include <SPI.h>
#include <SD.h>
#include <Ticker.h>

#ifdef DUMP_AT_COMMANDS  // if enabled it requires the streamDebugger lib
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, Serial);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

#define uS_TO_S_FACTOR 1000000ULL  // Conversion factor for micro seconds to seconds
#define TIME_TO_SLEEP  60          // Time ESP32 will go to sleep (in seconds)

#define UART_BAUD   115200
#define PIN_DTR     25
#define PIN_TX      27
#define PIN_RX      26
#define PWR_PIN     4

#define SD_MISO     2
#define SD_MOSI     15
#define SD_SCLK     14
#define SD_CS       13
#define LED_PIN     12

int counter, lastIndex, numberOfPieces = 24;
String pieces[24], input;

void setup()
{
    // Set console baud rate
    Serial.begin(115200);
    delay(10);

    // Set LED OFF
    pinMode(LED_PIN, OUTPUT);
    digitalWrite(LED_PIN, HIGH);

    pinMode(PWR_PIN, OUTPUT);
[... 241 lines cut ...]
    }

    Serial.println("\n---End of GPRS TEST---\n");
#endif

#if TINY_GSM_POWERDOWN
    // Try to power-off (modem may decide to restart automatically)
    // To turn off modem completely, please use Reset/Enable pins
    modem.sendAT("+CPOWD=1");
    if (modem.waitResponse(10000L) != 1) {
        DBG("+CPOWD=1");
    }
    modem.poweroff();
    Serial.println("Poweroff.");
#endif

    esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
    delay(200);
    esp_deep_sleep_start();

    Serial.println("This will never be printed");
}
I did not find for LILYGO® TTGO T-A7670E

https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_TinyGSM/AllFunctions/AllFunctions.ino

sumesh143, 2025-07-25:
Is this the link 

https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/lib/TinyGSM/examples/AllFunctions/AllFunctions.ino

lewisxhe, 2025-07-25:
Please read carefully how to get started.
https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX#4%EF%B8%8F%E2%83%A3-arduino-ide-quick-start

closed 2025-09-09 by github-actions[bot]