Xinyuan-LilyGO/LilyGo-Modem-Series #343: [Bug]: Error connecting to MQTTS EMQX

FacundoOR, 2025-08-14:
### Category

MQTT

### Hardware

T-A7670

### PN Number

S2-10AZT-Z32GV

### Firmware Version

```Shell
Manufacturer: SIMCOM INCORPORATED
Model: A7670G-LLSE
Revision: A124B01A7670M7
A7670M7_B01V01_240722
IMEI: 867284065507156
```

### Description

I can't get the module to connect to MQTTs. I've already updated the firmware version and the problem persists. I'm using the EMQX example https://github.com/Xinyuan-LilyGO/LilyGo-Modem-Series/tree/main/examples/MqttsBuiltlnEMQX

### Relevant log output

```Shell
*ATREADY: 1
.
AT
AT

OK

AT+CPIN?
AT+CPIN?
+CME ERROR: 13

*ISIMAID: "A0000000871004FF54F00189000001FF"

+CPIN: READY
SIM card online
Set network apn : igprs.claro.com.ar
AT+CGDCONT=1,"IP","igprs.claro.com.ar"

+CGEV: EPS PDN ACT 1

SMS DONE
AT+CGDCONT=1,"IP","igprs.claro.com.ar"

OK
Wait for the modem to register with the network.AT+CEREG?
AT+CEREG?

+CEREG: 0,1

OK
Online registration successful

Registration Status:1
AT+CPSI?
AT+CPSI?

+CPSI: LTE,Online,722-310,0x5410,1360140,485,EUTRAN-BAND7,3050,5,5,15,55,56,5

OK
Inquiring UE system information:LTE,Online,722-310,0x5410,1360140,485,EUTRAN-BAND7,3050,5,5,15,55,56,5
AT+NETOPEN
AT+NETOPEN

OK

+NETOPEN: 0AT+IPADDR

PB DONE
AT+IPADDR

+IPADDR: 10.53.184.113

OK
Network IP:10.53.184.113
AT+SIMCOMATI
AT+SIMCOMATI

Manufacturer: SIMCOM INCORPORATED
Model: A7670G-LLSE
Revision: A124B01A7670M7
A7670M7_B01V01_240722
IMEI: 867284065507156

OK
AT+SIMCOMATI

Manufacturer: SIMCOM INCORPORATED
Model: A7670G-LLSE
Revision: A124B01A7670M7
A7670M7_B01V01_240722
IMEI: 867284065507156

OK

AT+CMQTTSTART
AT+CMQTTSTART

OK

+CMQTTSTART: 0
Connecting to broker.emqx.ioAT+CCERTDOWN="ca_cert.pem",1339
AT+CCERTDOWN="ca_cert.pem",1339

>
-----BEGIN CERTIFICATE-----
MIIDrzCCApegAwIBAgIQCDvgVpBCRrGhdWrJWZHHSjANBgkqhkiG9w0BAQUFADBh
MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3
d3cuZGlnaWNlcnQuY29tMSAwHgYDVQQDExdEaWdpQ2VydCBHbG9iYWwgUm9vdCBD
QTAeFw0wNjExMTAwMDAwMDBaFw0zMTExMTAwMDAwMDBaMGExCzAJBgNVBAYTAlVT
MRUwEwYDVQQKEwxEaWdpQ2VydCBJbmMxGTAXBgNVBAsTEHd3dy5kaWdpY2VydC5j
b20xIDAeBgNVBAMTF0RpZ2lDZXJ0IEdsb2JhbCBSb290IENBMIIBIjANBgkqhkiG
9w0BAQEFAAOCAQ8AMIIBCgKCAQEA4jvhEXLeqKTTo1eqUKKPC3eQyaKl7hLOllsB
CSDMAZOnTjC3U/dDxGkAV53ijSLdhwZAAIEJzs4bg7/fzTtxRuLWZscFs3YnFo97
nh6Vfe63SKMI2tavegw5BmV/Sl0fvBf4q77uKNd0f3p4mVmFaG5cIzJLv07A6Fpt
43C/dxC//AH2hdmoRBBYMql1GNXRor5H4idq9Joz+EkIYIvUX7Q6hL+hqkpMfT7P
T19sdl6gSzeRntwi5m3OFBqOasv+zbMUZBfHWymeMr/y7vrTC0LUq7dBMtoM1O/4
gdW7jVg/tRvoSSiicNoxBN33shbyTApOB6jtSj1etX+jkMOvJwIDAQABo2MwYTAO
BgNVHQ8BAf8EBAMCAYYwDwYDVR0TAQH/BAUwAwEB/zAdBgNVHQ4EFgQUA95QNVbR
TLtm8KPiGxvDl7I90VUwHwYDVR0jBBgwFoAUA95QNVbRTLtm8KPiGxvDl7I90VUw
DQYJKoZIhvcNAQEFBQADggEBAMucN6pIExIK+t1EnE9SsPTfrgT1eXkIoyQY/Esr
hMAtudXH/vTBH1jLuG2cenTnmCmrEbXjcKChzUyImZOMkXDiqw8cvpOp/2PV5Adg
06O/nVsJ8dWO41P0jmP6P6fbtGbfYmbW0W5BjfIttep3Sp+dWOIrWcBAI+0tKIJF
PnlUkiaY4IBIqDfv8NZ5YBberOgOzW6sRBc4L0na4UU+Krk2U886UAb3LujEV0ls
YSEY1QSteDwsOoBrp+uvFRTp2InBuThs4pFsiv9kuXclVzDAGySj4dzp30d8tbQk
CAUw7C29C79Fv1C5qfPrmAESrciIxpg0X40KPMbp1ZWVbd4=
-----END CERTIFICATE-----

OK
AT+CSSLCFG="cacert",0,"ca_cert.pem"
AT+CSSLCFG="cacert",0,"ca_cert.pem"

OK
AT+CSSLCFG="sslversion",0,4
AT+CSSLCFG="sslversion",0,4

OK
AT+CMQTTSSLCFG=0,0
AT+CMQTTSSLCFG=0,0

OK
AT+CSSLCFG="authmode",0,1
AT+CSSLCFG="authmode",0,1

OK

AT+CMQTTREL=0
AT+CMQTTREL=0

+CMQTTREL: 0,20

ERROR
AT+CMQTTACCQ=0,"A76XX",1
AT+CMQTTACCQ=0,"A76XX",1

OK
AT+CMQTTCFG="version",0,4
AT+CMQTTCFG="version",0,4

OK
AT+CMQTTCONNECT=0,"tcp://broker.emqx.io:8883",60,1
AT+CMQTTCONNECT=0,"tcp://broker.emqx.io:8883",60,1
Failed!
Please make sure you are using the latest released version of the firmware. Find the latest version here: https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/blob/main/docs/update_fw.md
If you still have problems with the latest firmware, please open an issue. Otherwise, please do not create meaningless issues.
```

lewisxhe, 2025-08-15:
Please check your sim card.

FacundoOR, 2025-08-19:
Thank you very much @lewisxhe, problem solved. I'd also like to add that in order to connect to my private EMQX server, I had to enable SNI.
modem.sendAT("+CSSLCFG=\"enableSNI\",0,1"); // Enable SNI

closed 2025-10-04 by github-actions[bot]