Xinyuan-LilyGO/LilyGo-Modem-Series #344: SIM7670G auto-reset

Caanee, 2025-08-16:
Hello everyone, I’m having an issue with my LilyGO T-SIM7672G v1.1 with the SIM7670G modem.

The problem is that the modem restarts every so often. It looks like the ESP32 sends AT+CPIN? and then the modem resets.
Any help with this problem would be greatly appreciated.

I haven’t tested the SIM card yet, but it should be working fine.

Best regards,

Below I’m attaching the serial log:

I upload the Sketch AT-debug. 

The first is the firmware B03V02

```
22:26:59.414 -> AT+SIMCOMATI
22:26:59.446 -> Manufacturer: SIMCOM INCORPORATED
22:26:59.478 -> Model: SIM7670G-MNGV
22:26:59.510 -> Revision: 2374B03SIM767XM5A_M
22:26:59.542 -> SIM767XM5_B03V02_240914
22:26:59.542 -> IMEI: 864643060186410
```

After this is the log:
```

22:21:25.948 -> Start Sketch
22:21:25.948 -> Trying baud rate 115200
22:21:29.059 -> Modem responded at rate:115200
22:21:29.059 -> ***********************************************************
22:21:29.059 ->  You can now send AT commands
22:21:29.059 ->  Enter "AT" (without quotes), and you should see "OK"
22:21:29.059 ->  If it doesn't work, select "Both NL & CR" in Serial Monitor
22:21:29.059 ->  DISCLAIMER: Entering AT commands without knowing what they do
22:21:29.059 ->  can have undesired consiquinces...
22:21:29.059 -> ***********************************************************
22:21:29.059 -> 
22:21:33.735 -> �
22:21:33.922 -> QCRDY
22:21:34.316 -> 
22:21:34.361 -> +CPIN: READY
22:21:34.641 -> 
22:21:34.641 -> SMS DONE
22:21:44.146 -> >
�
22:21:44.348 -> QCRDY
22:21:44.780 -> 
22:21:44.780 -> +CPIN: READY
22:21:45.056 -> 
22:21:45.092 -> SMS DONE
22:21:51.435 -> 
22:21:51.668 -> QCRDY
22:21:52.073 -> 
22:21:52.073 -> +CPIN: READY
22:21:52.384 -> 
22:21:52.384 -> SMS DONE
```

Then I tried to upload the firmware:

```
22:30:43.634 -> Manufacturer: SIMCOM INCORPORATED
22:30:43.666 -> Model: SIM7670G-MNGV
22:30:43.698 -> Revision: 2374B05SIM767XM5A_M
22:30:43.730 -> SIM767XM5_B05V01_241206
22:30:43.730 -> IMEI: 864643060186410
```

It's the log:

```
22:32:12.234 -> Start Sketch
22:32:12.234 -> Trying baud rate 115200
22:32:15.250 -> Modem responded at rate:115200
22:32:15.250 -> ***********************************************************
22:32:15.250 ->  You can now send AT commands
22:32:15.250 ->  Enter "AT" (without quotes), and you should see "OK"
22:32:15.250 ->  If it doesn't work, select "Both NL & CR" in Serial Monitor
22:32:15.250 ->  DISCLAIMER: Entering AT commands without knowing what they do
22:32:15.250 ->  can have undesired consiquinces...
22:32:15.250 -> ***********************************************************
22:32:15.250 -> 
22:32:19.283 -> �
22:32:19.283 -> 
22:32:19.440 -> QCRDY
22:32:19.856 -> 
22:32:19.894 -> +CPIN: READY
22:32:20.172 -> 
22:32:20.172 -> SMS DONE
22:32:26.729 -> ]�
22:32:26.963 -> QCRDY
22:32:27.354 -> 
22:32:27.354 -> +CPIN: READY
22:32:27.629 -> 
22:32:27.662 -> SMS DONE
```

lewisxhe, 2025-08-16:
Did the modem light go off during the reboot? Then come back on? Or was it completely off?

Have you replaced the USB-C cable? A poor-quality cable won't provide enough current to connect to the network.

Caanee, 2025-08-16:
When I restart the modem, it goes through the correct reset sequence: the STATUS LED blinks quickly during reset, then it starts up.

After that, it stays in a strange sequence:
STATUS (about 1s) → NETLIGHT (about 1s) → both turn off → STATUS (about 1s) → NETLIGHT (about 1s) → both turn off again, repeating.

During that sequence, it prints the following log:

```
23:45:45.396 -> ~Z�
23:45:45.669 -> QCRDY
23:45:46.122 -> 
23:45:46.122 -> +CPIN: READY
23:45:46.400 -> 
23:45:46.436 -> SMS DONE
23:45:48.368 -> �
23:45:48.601 -> QCRDY
23:45:49.005 -> 
23:45:49.053 -> +CPIN: READY
23:45:49.331 -> 
23:45:49.364 -> SMS DONE
23:45:51.161 -> ��
23:45:51.422 -> QCRDY
23:45:51.837 -> 
23:45:51.872 -> +CPIN: READY
23:45:52.202 -> 
23:45:52.202 -> SMS DONE
```

I already tried with a good quality USB-C cable (it’s the official RPi5 5V 5A charger), so power delivery should not be the issue. It also has a battery connected.

I still need to test whether the SIM card is actually the problem, because when I remove it, the modem stays in standby (STATUS and NETLIGHT both on), which is the expected behavior when no SIM card is installed.

lewisxhe, 2025-08-16:
I don't think it's related to the SIM card. Detecting a SIM card will trigger the network connection process, which requires significantly more power than without a SIM card. I still suspect your power supply is the problem.
Connecting a battery won't improve power, as they operate on different lines. If you connect a USB, power will only come from that source.

Caanee, 2025-08-16:
If it were the power supply, that charger I mentioned has a power output of 25W, more than enough for the modem. In fact, something interesting that happened was that the PCBA was working perfectly until it entered that strange mode. I don’t know if some component may have failed, or if the modem’s flash might have been damaged.

In any case, I have more T-SIM7672G units in the office that are working right now, but this one I have at home went into that abnormal failure.

Has an issue like this ever happened before? I’ve never had this problem with the T-SIM7672G.

I also ended up performing a shutdown with the command: AT+CPOF. This shouldn’t have made the modem enter a strange mode, right?

In any case, is there a way to return the modem to factory settings and then update it again to B05V01?

Caanee, 2025-09-04:
Hi, I still need help with the Lily. I’ve run into a new issue: another Lily was working fine, but the modem suddenly lost connection and can’t register (two red light on blocked). It won’t register with any SIM—I’ve tried all 16 from the project.
Thanks!

closed 2025-10-19 by github-actions[bot]