Xinyuan-LilyGO/LilyGo-Modem-Series #351: [Bug]: SIM7670G SIM card missing

realbeeit, 2025-08-22:
### Category

Hardware

### Hardware

T-SIM7670G S3

### PN Number

S2-10D1Y-Z32W4

### Firmware Version

```Shell
SIMCOM INCORPORATED
Model: SIM7670G-MNGV
Revision: 2374B04SIM767XM5A_M
SIM767XM5_B04V01_241010
IMEI: 864643060182583
```

### Description

Device was working properly and suddenly it stopped. Only one red LED is on and on AT+CPIN? respond with SIM failure even if SIM is ok and tested. I tested with another SIM and another carrier. SIM is not the issue. It just stopped out of nowhere. We tried reset the modem, clear the memory, nothing helped. Board was working for 25 days, brand new and just stopped. It was enclosed in box. Could be dead modem board?

### Relevant log output

```Shell

```

realbeeit, 2025-08-22:
No, it respond with SIM card failure. I cleaned the sim tray and SIM card is 100% ok.

realbeeit, 2025-08-22:
I recorded the video but I can't upload it now. Both SIM cards say the same error. I will send the video to LilyGo.cc where I bought the board.

lewisxhe, 2025-09-05:
Just wait patiently.

realbeeit, 2025-09-17:
Another issue with brand new board, never connected.
14:14:17.389 > +CREG: 0,2
14:14:17.390 >
14:14:17.390 > OK
14:14:17.390 > AT+CSQ
14:14:17.396 > AT+CSQ
14:14:17.396 > +CSQ: 99,99
14:14:17.397 >
14:14:17.397 > OK
14:14:17.397 > [301] Signal Quality:99
14:14:18.397 > AT+CREG?
14:14:18.403 > AT+CREG?
14:14:18.404 > +CREG: 0,2
14:14:18.404 >
14:14:18.404 > OK
14:14:18.405 > AT+CSQ
14:14:18.410 > AT+CSQ
14:14:18.411 > +CSQ: 99,99
14:14:18.412 >
14:14:18.412 > OK
14:14:18.412 > [302] Signal Quality:99
14:14:19.412 > AT+CREG?
14:14:19.418 > AT+CREG?
14:14:19.419 > +CREG: 0,2

LEDS are constanly on and after couple of seconds went off and on again and in loop. Tried different antenna, different SIM. I am starting to think that these boards have some serious issues with networking.

lewisxhe, 2025-09-18:
What power supply are you using? I suspect the voltage isn't high enough.

realbeeit, 2025-09-18:
orders@lilygo.cc

lewisxhe, 2025-09-18:
Which email address did you contact?

IOTESLA-ing, 2025-09-22:
Dear Sir, We have the same problem , we bought 15 lilygo SIM7670 G , 02 with blocked modem, and 01 unit with restarting modem. actions : we have lowered the firmware version  and also the updated it to the latest version ,both times have been unsuccessful. We are very worried becasuse we are running lilygo on an important project.

lewisxhe, 2025-09-25:
I have seen your email and will contact you in the next two days.

lewisxhe, 2025-10-07:
> Failed to find INA3221 chip Serial monitor display this message when I upload this code

I'm referring to the parameters passed to Wire.begin instead of using this example.

https://github.com/Xinyuan-LilyGO/LilyGo-Modem-Series/blob/f67eed12e5bffee0114c0a011825a40687a6fafb/examples/SimShieldCurrentSensor/SimShieldCurrentSensor.ino#L7

As mentioned above, this is only suitable for use with SimShield.

lewisxhe, 2025-10-07:
> I need to know by AT commands how much data I use. And I know it depends on my network operator but is this module firmware supoort the USSD code? I try but no result! Thanks

Not supported