Xinyuan-LilyGO/LilyGo-Modem-Series #353: Solar power for LILYGO® TTGO T-A7670G/E

TomRec-eng, 2025-08-26:
Hi Lewis,

  Issue: LilyGO T-A7670G experiencing unexpected reboots during solar/battery power transitions.

  Expected behavior: Seamless power switching between solar and battery without any system interruptions or reboots.

  Actual behavior:
  - Device reboots every time solar power stops (clouds/night)
  - Device reboots again when solar power resumes
  - With intermittent solar (cloudy conditions), device gets stuck in continuous reboot loops
  - System never maintains continuous operation across power source transitions

  Technical details:
  - Battery voltage reads 4.41-4.42V (appears fully charged)
  - System correctly detects charging_mode = true/false
  - All reboots show reading_count reset to 0, confirming full system restart
  - Data transmission works perfectly when system is stable

  Questions:
  1. Is this normal behavior for the T-A7670G power management?
  2. Should the battery provide seamless backup power without any system interruption?
  3. Are there hardware jumpers, power management settings, or code configurations needed for continuous operation?
  4. Could this indicate a faulty battery connection or charging circuit?

  The sensor monitoring works great when stable - just need to eliminate these power transition reboots.

  Thanks,
  Tom

lewisxhe, 2025-08-27:
The solar port is only for charging the battery, there is no power switching issue. Do you use solar power as the power input for the board?

morgana2313, 2025-09-01:
I think this is the same issue as raised in #182 

And I think this is a design flaw you might want to improve on in a future hardware revision.

lewisxhe, 2025-09-02:
The new version has the function of seamless switching https://github.com/Xinyuan-LilyGO/LilyGo-Modem-Series#esp32s3-version-standard-series-get-started

lewisxhe, 2025-09-04:
This is a brand new series
[image]

morgana2313, 2025-09-04:
I found the [LilyGo T-SIM7670G-S3-Standard](https://github.com/Xinyuan-LilyGO/LilyGo-Modem-Series/blob/main/docs/en/esp32s3/sim7670g-s3-standard/REAMDE.MD). Is this an enhanced version of the [T-SIM7670G S3](https://lilygo.cc/products/t-sim-7670g-s3?srsltid=AfmBOor6WwOW3bQ49gC8BccdaGhip4cqqJ7fsbohKBVPwqCO_fRwW7FK) with new features:
* [schema](https://github.com/Xinyuan-LilyGO/LilyGo-Modem-Series/blob/main/schematic/standard/T-A7670X-S3-Standard_Rev1.0.pdf)
* QWIIC connector
* new power supply solution
* camera connector
* usb-c modem port
* PPS => gpio + led (nice!)

Do you have an image of the new board & formfactor?

lewisxhe, 2025-09-04:
The dimensions are exactly the same as before. The 3D has been stored at https://github.com/Xinyuan-LilyGO/LilyGo-Modem-Series/tree/main/dimensions/Standard

lewisxhe, 2025-09-04:
https://github.com/Xinyuan-LilyGO/LilyGo-Modem-Series/issues/293#issuecomment-3163128422

lewisxhe, 2025-09-04:
[image]
You can see that when the USB is disconnected, it does not affect the operation of the camera and modem.

closed 2025-10-25 by github-actions[bot]