Xinyuan-LilyGO/LilyGo-Modem-Series #354: Powering board from the "solar" connector and/or the 18650 LiPo battery.

JoanSalelles, 2025-08-28:
I can't get the board to work with power from the "solar" connector and the 18650 LiPo battery.
I am continuing to run tests with my SMS-based irrigation control project using the LilyGo T-A7670E board.
If I power the board via the USB-C port or by connecting 5V DC to Vbus, everything works perfectly. However, if I try to run the board powered by the battery and with or without power to the solar connector (for now I am not using a solar panel; I'm connecting the same 5V DC power supply used earlier for Vbus), nothing works correctly.

The first thing that happens is that when I power the board through the solar connector, it does not start immediately; I have to press the "reset" button, and then it starts up. I assume this must be the normal operation, right?

Once the board is running, the program configures the modem and sends an SMS. This SMS is very rarely sent correctly. On the few occasions that the SMS is sent, the modem is correctly left with the "Net Light" LED off and the "Status Light" LED solid. But if an SMS is sent to the board, the "Net Light" turns off and it seems like the modem is then offline.

I should note that if the board is powered by Vbus (using the same power supply, meaning it's not a problem of insufficient power from this source), upon receiving an SMS the "Status Light" LED blinks for a few seconds and then turns off, leaving only the "Net Light" on, and the SMS has been processed and replied to correctly, if applicable.

The same anomalous behavior is observed if the board is powered by USB-C or Vbus and then disconnected: with the LiPo battery fully charged at 4.2V, it is impossible to send or receive an SMS, and often it won't even register on the network.

Could you guide me on what might be happening and what further tests I could run to achieve correct operation with "solar" power and the battery, please?

Thank you very much in advance.

lewisxhe, 2025-08-29:
The solar power port is only for charging the battery, not for powering it. All power comes from the battery, not the solar port.

First, you need to confirm that the red indicator light on the modem is on when using the 4.2V battery. First, make sure the modem is powered on.

JoanSalelles, 2025-08-29:
Yes, when I power the board only from battery, not from USB-C nor VBus, after pressing the reset button of the board, the modem performs the correct starting cycle: 
- Status light goes on shortly for less tha a second
- After a few more seconds it lights again and remains on
- while the net light starts blinking
- Sometimes the SMS reporting the reset is sent and net light goes off, status remains on
- Everything is OH up to this point. But...
- When I sent back to the board an SMS from my phone, the status light goes off and the system will not respond again...
- Sometimes it doesn't even send the first SMS. Both LEDs turn off and the board stops responding.

lewisxhe, 2025-09-01:
Are you using our example or writing your own?
1. Is GPIO12 set high?
2. Is the USB power supply working properly?