Xinyuan-LilyGO/LilyGo-Modem-Series #356: SIM7670G - ATDebug failing to establish baud rate

tcranw, 2025-09-05:
Here is output from the ATDebug program. I've ran everything using PlatformIO:

Start Sketch
Wait modem started...
Trying baud rate 115200
Trying baud rate 9600
Trying baud rate 57600
Trying baud rate 38400
Trying baud rate 19200
Trying baud rate 74400
Trying baud rate 74880
Trying baud rate 230400
Trying baud rate 460800
Trying baud rate 2400
Trying baud rate 4800
Trying baud rate 14400
Trying baud rate 28800
***********************************************************
 Failed to connect to the modem! Check the baud and try again.
***********************************************************

The MqttsBuiltlnAWS program similarly fails, after watching the tutorial video and following its setup, it does not get past printing this message in a loop:
AT
AT
AT
AT
AT
.

I was given the board so I'm not sure if it was working previously or not, but I'm curious if any help can be offered. Thank you, and let me know any other useful information I can provide.

lewisxhe, 2025-09-09:
Record a video and upload it to the SIM7670G. Then open the serial port and wait for the message to appear on the serial port. Note that you must select #define LILYGO_T_SIM7670G_S3 in the utilities.h file. This must be reflected in the video and sent to the distributor.

tcranw, 2025-09-10:
> Record a video and upload it to the SIM7670G. Then open the serial port and wait for the message to appear on the serial port. Note that you must select #define LILYGO_T_SIM7670G_S3 in the utilities.h file. This must be reflected in the video and sent to the distributor.

Yes the issue was in the utilities.h file. I missed it.

closed 2025-09-10 by tcranw