Xinyuan-LilyGO/LilyGo-Modem-Series #358: SIM7670G SMS Only SIM Card + Firmware Upgrade Fix Issues

tcranw, 2025-09-10:
I was running the MqttsBuiltlnAWS program and encountered the following:

Registered for "SMS only", home network (applicable only when E-UTRAN), this type of registration cannot access the network. Please check the APN settings and ask the operator for the correct APN information and the balance and package of the SIM card. If you still cannot connect, please replace the SIM card and test again. Related ISSUE: https://github.com/Xinyuan-LilyGO/LilyGO-T-A76XX/issues/307#issuecomment-3034800353

I checked that issue and others, and determined I should upgrade the firmware on my board. Here is the AT+SIMCOMATI command response:
Manufacturer: SIMCOM INCORPORATED
Model: SIM7670G-MNGV
Revision: 2374B04SIM767XM5A_M
SIM767XM5_B04V01_241010
IMEI: 864643060179985

One issue thread from @lewisxhe states "The following firmware is ony suitable for SIM7670G P/N number S2-10D1Y-Z327P. Other modules cannot be flashed, otherwise it will become a brick." My board has P/N number of S2-10D1Y-Z32W4, and as seen in the above information, it is on software B04V01. There is an upgrade available for B05V01 that I want to test to see if it fixes the "SIM Only" issue, but now I'm unsure if that upgrade is compatible.

Additionally, I was having problems with the flashTools v4.0.7 being able to select my COM port. When selecting the Prj file and package, the COM ports disappear. The tutorial video and docs for the SIM7670G say that you should hold the SIM BOOT button near the micro-USB port, then plug in the micro-USB cable, then the USB-C cable, and wait for the COM to appear. I tried doing exactly as so, but when using the config_qcx216_usb, the COM fails to show up again after plugging in the board. All COM ports available in device manager (4,6,7,8,9) show up after plugging it in, and the COM4 port response to AT->OK over serial. There is a COM5 that does not show up on device manager but does show up when I load the config_qcx216_uart, however it does not work to flash over that port as it doesn't open (which makes sense, as I don't believe it exists). I am running the flashTools.exe on Windows 11.

Let me know any information you can provide. Thank you!

lewisxhe, 2025-09-13:
Windows 11 is OK, the operating system I used to record the video is win11.

lewisxhe, 2025-09-16:
I need a detailed network access process. You'll need to capture the modem's logs. A video is available here. Please follow the instructions to operate the log capture tool.

https://github.com/Xinyuan-LilyGO/LilyGo-Modem-Series/blob/main/docs/sim7670g_capture_log.md

What you need to do:
1. Insert your SIM card
1. Write to [ATdebug.ino](https://github.com/Xinyuan-LilyGO/LilyGo-Modem-Series/blob/main/examples/ATdebug/ATdebug.ino)
2. Plug in the Micro-USB and USB-C port
3. Use the tool to enable the logging port, as shown in the video
4. Restart the modem by sending AT+CFUN=6 via serial.
5. Then, repeatedly send AT+CREG? This command should be sent in a loop until CREG:1 or CREG:6 is displayed. You will need to record the logs as the modem connects to the network. When CREG:1 or CREG:6 is displayed, you can disable log capture and export the logs to send to me.

Only those who encounter this problem can know the cause of this problem by providing logs, because we cannot simulate this kind of problem.

closed 2025-11-22 by github-actions[bot]

IgorVolp, 2026-09-03:
I'm looking for the latest SIMCOM SIM7670G firmware relase. Could you suggest me a link to download it? Thank you

learhy, 2026-09-13:
@tcranw glad the Win10 machine worked — for what it's worth, you don't need Windows at all for this.

I flashed a SIM7670G-MNGV from macOS using an open-source tool (built on openLuat's `ectool`, which implements the EC618 USB download protocol in pure Python): https://github.com/learhy/sim767x-mac-flasher

Works on macOS and Linux. Two things that matter:

1. Hold the modem's BOOT button + power cycle so it enumerates as "QCom Boot" (05c6:9330) — a single serial port.
2. If the module has EVER been flashed before, it rejects the standard DLBOOT handshake (replies `0x2b02d3ee` instead of echoing) because its boot ROM auto-runs the resident agentboot. SIMCom's official QDL tool also fails on this — it's not your PC, it's the handshake state. The flasher above detects it and burns via the LPC phase directly. ~90 s for the whole package.

Verified post-flash via AT commands (CGMR/CSUB report the new version).