Xinyuan-LilyGO/LilyGo-Modem-Series #369: [Bug]: Sensor SHT21/HTU21

MatiasRaya, 2025-09-24:
### Category

Other

### Hardware

T-SIM7670G S3

### PN Number

S2-10D1Y-Z327P

### Firmware Version

```Shell
SIM767XM5_B05V01_241206
```

### Description

Hello, is it possible that you could share with me which pins I can use to connect the SHT21/HTU21 sensor since I am trying to connect it via i2c and it is not being feasible?

### Relevant log output

```Shell

```

lewisxhe, 2025-09-25:
Did you specify IO to initialize?
https://github.com/Xinyuan-LilyGO/LilyGo-Modem-Series/blob/b1cb0ff8496c71c64a6cec5abb4ea1a92fefe743/examples/SimShieldCurrentSensor/SimShieldCurrentSensor.ino#L24

MatiasRaya, 2025-09-25:
Hello, I initialized it on pin 1 and 2 of the plate. Still, it throws wrong readings at me. I would recommend how to do it?

lewisxhe, 2025-09-25:
Did you add the I2C pull-up resistors?

MatiasRaya, 2025-09-25:
Do not add pull-up resistors in SDA or SCL. in the same i2C, but with another address I have an RTC DA3231. How much does the resistance to add have to be?

MatiasRaya, 2025-09-25:
Yes, the DS3231 works correctly. as I can see if it has pull-up resistances in SDA and SCL. the link to the board I use is this 
https://es.aliexpress.com/i/1005005926933268.html

lewisxhe, 2025-09-25:
You can't see the resistor in the picture, so you can just connect the SDA and SCL pins of the DS3231 in parallel with the SHT21.

lewisxhe, 2025-09-26:
Paste your test code

lewisxhe, 2025-09-26:
```c
#include <Wire.h>
#include "Adafruit_HTU21DF.h"

#define SDA_PIN 1
#define SCL_PIN 2

Adafruit_HTU21DF htu = Adafruit_HTU21DF();

void setup()
{
    Serial.begin(115200);
    Serial.println("HTU21D-F test");

    Wire.begin (SDA_PIN, SCL_PIN);

    if (!htu.begin()) {
        Serial.println("Couldn't find sensor!");
        while (1);
    }
}

void loop()
{
    float temp = htu.readTemperature();
    float rel_hum = htu.readHumidity();
    Serial.print("Temp: "); Serial.print(temp); Serial.print(" C");
    Serial.print("\t\t");
    Serial.print("Humidity: "); Serial.print(rel_hum); Serial.println(" \%");
    delay(500);
}

```

# platformio.ini

```bash
lib_deps = 
  adafruit/Adafruit HTU21DF Library @ ^1.1.2
```